arduino-libraries/MKRWAN #3: LoraSendAndReceive: does not use parsePacket()

sandeepmistry, 2017-12-04:
The `LoraSendAndReceive` uses `available()` to check if there is a response after send a packet, is there any reason `parsePacket()` is not used in the example?

sslupsky, 2018-10-30:
I have a similar question.  The documentation on the web site says

> Checks for the presence of a LoRaWAN packet, and reports the size. parsePacket() must be called before reading the buffer with modem.read().

But the example uses available().  So, can you use one or the other function?  Is there a reason to use parsePacket instead of available()?

matthijskooijman, 2022-02-09:
I was wondering the same thing looking at the examples. I *thought* that maybe `parsePacket()` would return the size of a single packet (i.e. when two packets are received before the first one is fully read), but looking at the code it seems that `parsePacket()` just returns `available()` and multiple received packets just get appended to the same buffer, so there really is no way to distinguish packets.

Of course, when running in Class A mode, you can only get a single downlink packet after each uplink, so having two packets in the buffer at the same time can probably be avoided usually (until support for Class C is added, of course).