arduino-libraries/MKRWAN #7: Problem joining using ABP

xoseperez, 2018-01-26:
Version: 1.02
Device: Arduino MKR WAN 1300
Code: FirstConfiguration.ino example

The FirstConfiguration example does not work when using ABP.

After some debugging, I found out that the set(DEV_ADDR, devAddr); in joinABP returns false always. Going a little deeper: the AT+DEVADDR=XXXX command times out in the waitResponse. After inspecting code from STM32 Cube Expansion LoRaWan I tried using AT+DADDR but got the same result.

OTAA works great.

facchinm, 2018-01-29:
Hi @xoseperez ,
I analyzed the bug and, indeed, the firmware is behaving incorrectly in case of ABP registration.
I just pushed an updated firmware on https://github.com/arduino/mkrwan1300-fw/releases/tag/1.1.3  and I'm publishing the new library soon. Let me know if it fixes the issue for you!

xoseperez, 2018-01-29:
That link goes to a 404. That repo does not exist, or maybe it's not public.

facchinm, 2018-01-29:
It's public now, and it should have been since the beginning... However, you don't need that repo, simply update the libary (cloning `master`, I'm waiting for your feedback to release properly) and run `MKRWANFWUpdate_standalone` sketch

xoseperez, 2018-01-29:
Cool, module version ARD-078 1.1.3 up and working. 

Just one thing: out of the three tests I have done one gave an "Error sending message :(" but the message actually went through. Maybe the ACK?

facchinm, 2018-01-29:
:+1: 
could you also confirm that OTAA is still working good?
About the failing ACK, probably the timeouts are a bit too tight, I should increase them to avoid returning false negatives.

xoseperez, 2018-01-29:
OTAA is working too.
Thank you very much!

closed 2018-02-05 by facchinm