arduino-libraries/MKRWAN #11: FirstConfiguration doesn't work on 1.1.4

ldesomer, 2018-02-22:
Hi,
First Configuration sketch doesn't work on fw 1.1.4. Whatever the value you set to appKey and appEui the int "connected" will be equal to 1, even if you're not connected. Same problem when using ABP instead of OTAA.

facchinm, 2018-02-22:
Hi @ldesomer ,
I just tested it, and if I enter bogus data I always get 
```
Error sending message :(
```
but `connected()` returns true. While it's normal in ABP mode, it shouldn't in OTAA.
The FW only returns the joined event when the state machine enters `DEVICE_STATE_JOINED` status, so there is something wrong there

facchinm, 2018-02-22:
Fixed by 05f9804, thanks for reporting!

closed 2018-02-22 by facchinm

ldesomer, 2018-02-23:
Thanks for your quick reply and your help @facchinm !