arduino-libraries/MKRWAN #12: Spreading factor ?

ldesomer, 2018-03-02:
Hi,
Which SF does the library support ? Is it possible to modify it ?
Thanks for your help.

facchinm, 2018-03-09:
Hi @ldesomer,
the underlying firmware ( https://github.com/arduino/mkrwan1300-fw ) doesn't provide this yet, but we would be glad to receive pull requests :wink:

ldesomer, 2018-03-09:
Hi @facchinm,
Thanks for your reply, I'll try to work on it... How is the SF chosen by the firmware ? Or is it fixed ? :)

facchinm, 2018-03-09:
It looks like the spreading factor is only set during RF_TEST (https://github.com/arduino/mkrwan1300-fw/blob/ed493cec30273953e92ccdd210f2ef4240ef35fa/Projects/Multi/Applications/LoRa/AT_Slave/src/test_rf.c#L98), I don't know the LoRAWAN standard well enough to know if it can be changed (if not running point-to-point)

facchinm, 2018-06-28:
Probably there is a way to do it by changing the LoRaWAN stack on the firmware but it's quite a big effort.

matthijskooijman, 2022-05-24:
What you can do, is set the data rate (DR) using the `dataRate()` method. This configures one of the DR's from the LoRaWAN spec, each of which uses a specific SF. This does not allow choosing arbitrary or non-standard SF's, but is the normal way of configuring the DR/SF to use.