arduino-libraries/MKRWAN #13: Sleep mode ?

TMesot, 2018-03-03:
Hello, 
I notice the sleep function sending the SLEEP AT command but it doesn't seems to work. Is there any documentation about the sleep/idle/low power part ?

Thanks

facchinm, 2018-03-05:
Hi @TMesot ,
the low power handling is automatically performed by the module when a transaction (or a radio event) is not in progress (see https://github.com/arduino/mkrwan1300-fw/blob/master/Projects/Multi/Applications/LoRa/AT_Slave/src/main.c#L294). AT+SLEEP was implemented in Murata closed-source firmware and is kept for compatibility reasons; we'll eventually deprecate the unused functions if we'll never ship a board with that firmware.

TMesot, 2018-03-06:
Hello, 
Oh perfect thanks, I didn't notice the firmware was open-source, that's great !

closed 2018-03-06 by TMesot