arduino-libraries/MKRWAN #21: FSB (Frequency Sub Band) on MKRWAN1300

almarion, 2018-06-21:
I am pretty sure that the Murata module has AT command support for this, but it does not seem to be an available choice in the Arduino sketches.  If I look at the MKRWAN.h library, it would seem to be lacking as well.  This feature would allow most of us using US915 mode to select a sub band of 8 channels out of 64. Most mid-price gateways such as Multitech and Laird work this way.
Thanks for the community support.

almarion, 2018-07-06:
As a compromise, the hybrid choice gives the equivalent of FSB=1 i.e. uses the bottom 8 channels.

// Select your region (AS923, AU915, EU868, KR920, IN865, US915, US915_HYBRID)
_lora_band region = US915_HYBRID;

sslupsky, 2018-09-16:
Hi @almarion Can you please point me to where the channels are defined for the US915_HYBRID.  By trial and error I have concluded that channels 0 - 7 are used.  I have been unsuccessful in tracking down where its actually defined in the source code.  I compared the region header files for the US915 and US915_HYBRID and they are the same.