arduino-libraries/MKRWAN #23: How to p2p btw two MKRWAN boards ?

yijenlu1971, 2018-07-17:
I have two arduino MKRWAN 1300, and want to send & receive by point to point. One board I loaded DumbModemLoRaSender, the other board I loaded LoRaReceiver. But, the receiver can't get anything. Please give me help, thanks.

facchinm, 2018-07-18:
Hi @yijenlu1971 ,
using @sandeepmistry 's LoRa library (https://github.com/sandeepmistry/arduino-LoRa) version 0.5.0 you already have everything you need; launch `LoRaReceiver` on one board and `LoRaSender` on the other and it should be already working.
Two tips: remove the boards from the foam, otherwise there could be little short cicuits between the pins that make it behave "randomly".
Then, make sure the firmware version of LoRa modules is up to date (using  `MKRWANFWUpdate_standalone` example).
Let me know if it works!

yijenlu1971, 2018-07-18:
Thanks your suggestion. It works now.
Try to debug. I put LED indicator to LoRaSender. When it send out, LED will flush. So, I found the board's not running at standalone. Now, both boards are loaded before running. It seems UART is a trigger to make arduino-mkrwan run or wakeup, is that right ??

[image]

closed 2018-07-18 by yijenlu1971

facchinm, 2018-07-19:
@yijenlu1971 I'm happy it works! The serial "trigger" is due to this line

https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiver/LoRaReceiver.ino#L6

that stops the execution until you open the serial port

yijenlu1971, 2018-07-19:
Thanks very much.

bradtothebone, 2018-11-26:
same thing but I get: Starting LoRa failed!

I tried checking the freq was correct (in lorawan more this needs to be US916):
if (!LoRa.begin(915)) {
    Serial.println("Starting LoRa failed!");

I did update the firmware with the standalone sketch too, maybe incorrectly? I get version 2 something and the directions say min is 1.1.6:

Any ideas? Thanks

Device       : ATSAMD21G18A
Chip ID      : 10010005
version()=v2.0 [Arduino:XYZ] Nov 13 2017 18:12:39
Version      : v2.0 [Arduino:XYZ] Nov 13 2017 18:12:39

facchinm, 2018-11-26:
Hi @bradtothebone ,
you need to open the serial monitor after flashing the standalone updater for it to apply.
Then, using LoRa library >= 0.5.0 you'll be able to talk with the chip.

bradtothebone, 2018-11-26:
Thanks.. I basically mixed up which chips I had flashed. few of them on the desk, one on a breadboard as a module. phew, I thought I had killed it by having the antenna off!

rafaelcastellar, 2019-04-07:
Hi guys!

I'm stuck here with my two MKRWAN1300 p2p. I did what you said here, but I'm still getting the "Starting LoRa failed!" error.

I had updated the firmware from sandeepmistry sketch (MKRWAN has mlm32l07x01_bin_len error - not declared in the scope)
I'm running sandeepmistry Sender and Receiver sketches, both are giving me this messages.

A dumb question: the devices need to be wired? I'm just trying them wired in nothing but the usb cable to my computer.
The "First config" example from MKRWAN works fine, so I thought it is not a hardware problem..anyway, I don't know how to have them working.

Any help?

Tks!!

facchinm, 2019-04-08:
The error while compiling FwUpdater is not a good sign, are you sure you have downloaded the whole MKRWAN library? Once updated, there's no special wiring to do, just use Sandeep's library (at least version 0.5) and you should be ok.

rafaelcastellar, 2019-04-08:
Hi FAcchinm!
I was able to update just with Sandeep's sketch, the erro variable scope erro I was getting from MKRWAN library. But you were right, something was missing. I removed it and downloaded again and now this updated worked as Sandeep's: 
```
13:06:20.956 ->  Wrote and verified address 0x08012b00 (99%)
13:06:21.023 ->  Wrote and verified address 0x08012c00 (99%)
13:06:21.090 ->  Wrote and verified address 0x08012d00 (100%)
13:06:21.156 ->  Done.
13:06:21.156 -> 
13:06:21.156 -> Starting execution at address 0x08000000... done.
13:06:21.156 -> Flashing ok :)
13:06:22.022 -> ARD-078 1.1.9
```
 But when I run the Sandeep's LoRaSender.ino I still get this error:
```
13:10:18.003 -> LoRa Sender
13:10:18.454 -> Starting LoRa failed!
```
I'm pretty sure I am doing something wrong once it happens with my two boards.
Tks

superkalt, 2019-04-08:
hi,
firmwware  1.1.9  work with this sketch =>

`#include <MKRWAN.h>

LoRaModem modem;

// Mode 1 = OTAA
// Mode 2 = ABP

int ttnMode = 2 ;
boolean ttnCheck = false;

int ttnChannel = 1;
int ttnPort = 1;
int ttnDR = 1;

char *appEui = "xx";
char *appKey = "xx";

char *devAddr = "xx";
char *nwkSKey = "xx";
char *appSKey = "xx";

//String appEui = "xx";
//String appKey = "xx";
//String devAddr = "xx";
//String nwkSKey = "xx";
//String appSKey = "xx";

void setup() {

  Serial.begin(115200);
  //while (!Serial);
  delay(2000);
  delay(random(1000, 2000));

  Serial.println("TTN Mapper fuer Muenchen");
  Serial.println("Device: MUMAP-00");

  // Starte LoRa Modem

  //Serial.println("Reset LoRa Modem");

  //modem.factoryDefault();
  //modem.changeClass(CLASS_A);

  if (!modem.begin(EU868)) {
    Serial.println("Fehler beim start LoRa Modem");
    while (1) {}
  }

  // auslesen der wichtigsten Daten

  Serial.print("Version: ");
  Serial.println(modem.version());
  Serial.print("Device EUI: ");
  Serial.println(modem.deviceEUI());

  //Serial.print("DataRate: ");
  //Serial.println(modem.getDataRate());
  //Serial.print("FCU: ");
  //Serial.println(modem.getFCU());
  //Serial.print("FCD: ");
  //Serial.println(modem.getFCD());

  // Verbinden zu TTN

  int connected;
  if (ttnMode == 1) {
    Serial.println("Verbindung mit OTAA Mode");
    devAddr = "";
    nwkSKey = "";
    appSKey = "";
    connected = modem.joinOTAA(appEui, appKey);
  } else if (ttnMode == 2) {
    appEui = "";
    appKey = "";
    Serial.println("Verbindung mit ABP Mode");
    connected = modem.joinABP(devAddr, nwkSKey, appSKey);
  }

  if (!connected) {
    Serial.println("Keine Verbindung zum TTN Netzwerk moeglich");
    Serial.println("Prgramm Abbruch > Endlosschleife");
    while (1) {}
  } else if (connected) {
    Serial.println("Verbindung zum TTN Netzwerk hergestellt");
  }

  // Pause 5 Sekunden

  delay(5000);

}

// Start loop

void loop() {

  Serial.println("Start loop");

  // in 0-5 DataRate senden

  for (int i = 0; i <= 6; i++) {

    ttnDR = i;
    ttnChannel = 1;
    ttnPort = 3;

    Serial.print("Loop:");
    Serial.println(i);

    String payload = "D:" + String(ttnDR) + "|C:" + String(ttnChannel) + "|P:" + String(ttnPort);
    Serial.println(payload);

    Serial.println("Daten senden");

    if (ttnMode == 2) {
      Serial.println("ABP => Anpassung moeglich");
      //modem.setPort(ttnPort);
      //modem.dataRate(ttnDR);
    } else if (ttnMode == 1) {
      Serial.println("OTAA => keine Anpassung moeglich");
    }

    int err = 0;

    modem.beginPacket();
    modem.print(payload);
    //int err = modem.endPacket(true);
    err = modem.endPacket();

    if (ttnCheck) {

      Serial.println("Auswertung der Fehler");

      if (err > 0) {
        Serial.print(err);
        Serial.println(" > Nachricht gesendet");
      } else {
        Serial.print(err);
        Serial.println(" > Fehler beim senden");
      }

      // Daten empfangen

      Serial.println("Daten empfangen nach 1 Sekunde Pause");

      delay(1900);
      if (!modem.available()) {
        Serial.println("Keine Daten erhalten");
      } else {
        Serial.println("Daten lesen und ausgeben");
        char rcv[64];
        int g = 0;
        while (modem.available()) {
          rcv[g++] = (char)modem.read();
        }
        Serial.print("Empfangen: ");
        for (unsigned int j = 0; j < g; j++) {
          Serial.print(rcv[j] >> 4, HEX);
          Serial.print(rcv[j] & 0xF, HEX);
          Serial.print(" ");
        }
        Serial.println();
      }
    }

    // Pause 600 Sekunden

    Serial.print("Pause 120 Sekunden");
    for (int w = 0; w <= 120 / 5 ; w++) {
      delay(5000);
      Serial.print(w * 5);
      Serial.print("|");
    }
    Serial.println("");
  }
}`

rafaelcastellar, 2019-04-08:
I Superkalt! I will try it later, right now I'm not covered by any gateway.

Tks

rafaelcastellar, 2019-04-09:
Hi Superkalt. I tried but it seems not ok. Don't know if I set everything right. I used the correct keys created in TTN and I should be covered by a gate right now.
That's the logs:
```
21:43:10.303 -> TTN Mapper fuer Muenchen
21:43:10.303 -> Device: MUMAP-00
21:43:11.131 -> Version: ARD-078 1.1.9
21:43:11.230 -> Device EUI: a8610a30373c6001
21:43:11.230 -> Verbindung mit ABP Mode
21:43:11.330 -> Verbindung zum TTN Netzwerk hergestellt
21:43:16.301 -> Start loop
21:43:16.301 -> Loop:0
21:43:16.301 -> D:0|C:1|P:3
21:43:16.301 -> Daten senden
21:43:16.334 -> ABP => Anpassung moeglich
21:43:16.334 -> Auswertung der Fehler
21:43:16.334 -> 11 > Nachricht gesendet
21:43:16.334 -> Daten empfangen nach 1 Sekunde Pause
21:43:18.355 -> Keine Daten erhalten
21:43:18.355 -> Pause 120 Sekunden0|5|10|15|20|25|30|35|40|45|50|55|60|65|70|75|80|85|90|95|100|105|110|115|120|
21:45:23.280 -> Loop:1
21:45:23.280 -> D:1|C:1|P:3
21:45:23.280 -> Daten senden
21:45:23.280 -> ABP => Anpassung moeglich
21:45:23.321 -> Auswertung der Fehler
21:45:23.321 -> 11 > Nachricht gesendet
21:45:23.321 -> Daten empfangen nach 1 Sekunde Pause
21:45:25.302 -> Keine Daten erhalten
21:45:25.302 -> Pause 120 Sekunden0|5|10|15|20|25|30|35|40|45|50|55|60|65|70|75|80|85|90|95|100|105|110|115|120|
21:47:30.231 -> Loop:2
21:47:30.231 -> D:2|C:1|P:3
21:47:30.231 -> Daten senden
21:47:30.231 -> ABP => Anpassung moeglich
21:47:30.231 -> Auswertung der Fehler
21:47:30.231 -> 11 > Nachricht gesendet
21:47:30.231 -> Daten empfangen nach 1 Sekunde Pause
21:47:32.252 -> Keine Daten erhalten
21:47:32.252 -> Pause 120 Sekunden0|5|10|15|20|25|30|35|40|45|50|55|60|65|70|75|80|85|90|95|100|105|110|115|120|
21:49:37.154 -> Loop:3
21:49:37.154 -> D:3|C:1|P:3
21:49:37.154 -> Daten senden
21:49:37.154 -> ABP => Anpassung moeglich
21:49:37.187 -> Auswertung der Fehler
21:49:37.187 -> 11 > Nachricht gesendet
21:49:37.187 -> Daten empfangen nach 1 Sekunde Pause
21:49:39.176 -> Keine Daten erhalten
21:49:39.176 -> Pause 120 Sekunden0|5|10|15|20|25|30|35|40|45|50|55|60|65|70|75|80|85|90|95|100|105|110|115|120|
21:51:44.094 -> Loop:4
21:51:44.094 -> D:4|C:1|P:3
21:51:44.094 -> Daten senden
21:51:44.094 -> ABP => Anpassung moeglich
21:51:44.127 -> Auswertung der Fehler
21:51:44.127 -> 11 > Nachricht gesendet
21:51:44.127 -> Daten empfangen nach 1 Sekunde Pause
21:51:46.121 -> Keine Daten erhalten
21:51:46.121 -> Pause 120 Sekunden0|5|10|15|20|25|30|35|40|45|50|55|60|65|70|75|80|85|90|95|100|105|110|115|120|
21:53:51.037 -> Loop:5
21:53:51.037 -> D:5|C:1|P:3
21:53:51.037 -> Daten senden
21:53:51.037 -> ABP => Anpassung moeglich
21:53:51.070 -> Auswertung der Fehler
21:53:51.070 -> 11 > Nachricht gesendet
21:53:51.070 -> Daten empfangen nach 1 Sekunde Pause
21:53:53.059 -> Keine Daten erhalten
21:53:53.059 -> Pause 120 Sekunden0|5|10|15|20|25|30|35|40|45|50|55|60|65|70|75|80|85|90|95|100|105|110|115|120|
```
Should I specify any code related to the gate? I should be covered by this gate: eui-b827ebfffede01eb from this https://www.thethingsnetwork.org/community/saopaulo/ here in São Paulo, Brazil.

Is this log enough to determine if there are problems with the boards to get them working in P2P mode?

Tks agains!

superkalt, 2019-04-09:
it looks fine. Please check the OTAA Mode. if this Mode is working the Board and the cover is ok. 
you need a script with english logging?

rafaelcastellar, 2019-04-09:
Hi Stefan! It did not connected in OTAA mode. Could it be a gateway problem? 
```
20:03:38.640 -> TTN Mapper fuer Muenchen
20:03:38.640 -> Device: MUMAP-00
20:03:39.468 -> Version: ARD-078 1.1.9
20:03:39.501 -> Device EUI: a8610a30373c6001
20:03:39.501 -> Verbindung mit OTAA Mode
20:04:40.517 -> Keine Verbindung zum TTN Netzwerk moeglich
20:04:40.517 -> Prgramm Abbruch > Endlosschleife
```
In the ABP mode it did not get the data sent, neither these data can be seen at TNN. 

Meanwhile, with LoRa.h it keeps 
Any tip?

superkalt, 2019-04-10:
:) ok. 
1. no TTN Network
2. broken Hardware
3. wrong keys

rafaelcastellar, 2019-04-10:
1. no TTN Network: may be, I'll try to reach the responsible for it here in São Paulo
2. broken Hardware: don't know, it begins with MKRWAN.h, just with LoRa.h it failed to begin
3. wrong keys: checked them several times, but will do again.

Vielen danke!

superkalt, 2019-04-10:
write a email .... superkalt is easy to find in the www

7n7n7n, 2019-04-11:
refaelcasteller !
I have exactly the same problem with you!   I hope to solve our problem quickly.
If you solved it, could you post about what you did?

and I'll do that too!

7n7n7n, 2019-04-11:
refaelcasteller !
I have exactly the same problem with you! I hope to solve our problem quickly.
If you solved it, could you comment about what you did?

and I'll do that too!

7n7n7n, 2019-04-11:
refaelcasteller !
I have the same problem with you! I hope to solve our problem quickly.
If you solved it, could you comment about what you did?

and I'll do that too!

sandeepmistry, 2019-04-11:
@bradtothebone @rafaelcastellar @7n7n7n SAMD Core v1.6.21 introduced a regression.

I've opened a pull request to fix it: https://github.com/arduino/ArduinoCore-samd/pull/400

For the short term I suggest downgrading to SAMD Core v1.6.20. in the board manager.

rafaelcastellar, 2019-04-13:
Thank you very much @sandeepmistry ! That worked!

One of them is sending packets, the other one is not getting them yet, it is freeze at message:
`20:30:23.293 -> LoRa Receiver`
But there is not "failed" message!

Tks!

Michael-List, 2019-04-17:
Downgrading to SAMD Core v1.6.20 works for me with the Arduino MKR 1300

mcradit, 2019-04-22:
Downgrading to SAMD Core v1.6.20 worked for me too with the Arduino MKR 1300

jetsonworld, 2020-07-31:
In terms of LoRa P2P test of Sandeep Mistry, I did both SAMD Core v1.6.20 and LoRa by Sandeep Mistry 0.5.0 but they still stuck like below that.  13:19:10.335 -> LoRa Receiver

LEONWA02, 2020-08-03:
Just try to change the MHz from 915E6 to 915 after this it worked for me! 
Why? I don't know just give it a try. ; )