arduino-libraries/MKRWAN #25: Poll interval documentation

cimm, 2018-07-21:
The [LoraSendAndReceive](https://github.com/arduino-libraries/MKRWAN/blob/master/examples/LoraSendAndReceive/LoraSendAndReceive.ino#L40) example sketch uses the `minPollInterval()` function but I can't seem to find any documentation for this function.

```c
modem.minPollInterval(60);
// NOTE: independently by this setting the modem will
// not allow to send more than one message every 2 minutes,
// this is enforced by firmware and can not be changed.
```

This only sets the `pollInterval` variable if I read the [source](https://github.com/arduino-libraries/MKRWAN/blob/4b1b97791dbdf42dcf007dec6591a85c261cfbb3/src/MKRWAN.h#L495) correctly so why does the example set it to 60 seconds when the comment underneath states it can't be less than 2 minutes? Shouldn't it be minimum 120 seconds?

Thank you!

facchinm, 2018-08-20:
Hi @cimm ,
`minPollInterval` was added to wrap an AT command from original Murata firmware; as you correctly noticed it is currently unused and can be removed.

matthijskooijman, 2022-05-24:
> minPollInterval was added to wrap an AT command from original Murata firmware; as you correctly noticed it is currently unused and can be removed.

I'm a bit confused, as the `pollInterval` variable (set byy `minPollInterval`) *does* seem to be used here: https://github.com/arduino-libraries/MKRWAN/blob/2ca83989f6571a39b06f08f324918cfc1720f0e1/src/MKRWAN.h#L657

Or did you mean that the entire `poll()` function can be removed? Its function is not documented, but it looks a bit like it is intended to check network connectivity by sending a dummy (but not entirely empty) confirmed uplink?

matthijskooijman, 2022-05-24:
>  Its function is not documented, but it looks a bit like it is intended to check network connectivity by sending a dummy (but not entirely empty) confirmed uplink?

Though it seems the `poll()` function does not return the status of the confirmed upllink, so maybe it is intended for something else. I initially thought it would send an uplink to give the network a chance to send a downlink (i.e. poll for a pending downlink), but then it makes no sense that it uses a confirmed uplink (though maybe it does so to ensure blocking until the RX window is completed and also knowing for sure whether a message is pending or not)?