arduino-libraries/MKRWAN #26: Set channel not implemented?

Michael-List, 2018-07-23:
I've build a single channel lora gateway. So my problem is that the Arduino MKR 1300 iterates over the different channels that can be found here(https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html). Is there a way to set a channel?

Michael-List, 2019-04-26:
Could solve my problem with the following library: https://github.com/sandeepmistry/arduino-LoRa

closed 2019-04-26 by Michael-List

deanpretorius, 2019-07-21:
Hi @Michael-List 
Could you please explain how you managed to set the channel on the MKR WAN 1300?

Michael-List, 2019-07-22:
I used this library: [arduino-LoRa](https://github.com/sandeepmistry/arduino-LoRa). I have an example project which contains this file: [arduino_lora_weatherstation](https://github.com/Michael-List/arduino_lora_weatherstation/blob/master/arduino_lora_weatherstation.ino)

In line 11 I set the frequency/channel:

`static int frequency = 868300000;`

Frequency 868 with channel 2 -> See [Frequency plans](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html)  (2. 868.3 - SF7BW125 to SF12BW125 and SF7BW250)

If you have further questions feel free to ask me

deanpretorius, 2019-07-29:
Hi Michael
I managed to set the Spreading Factor using the library that you suggested. But currently I am still facing a problem with the frequency hopping of the MKRWAN. Please email me at deanpretorius@outlook.com
I have a few more questions that I would like to ask.

Michael-List, 2019-07-30:
Hi deanpretorius,
did you use the MKRWAN or arduino-LoRa library for sending data?

deanpretorius, 2019-07-30:
arduino-LoRa