arduino-libraries/MKRWAN #27: US915 support

conradholt, 2018-07-26:
Has someone been able to configure the mkr wan 1300 using the MKRWAN library on the us915 band? I'm not able to see any packet going through the gateway (This is configured with the 915E6 band as well) whereas if using the 868 band I can. 
What kind of gateway and single_channel_packet are you using? 
Is there any place with a detailed explanation and examples of the AT commands for the MURATA module? f.e when setting AT+BAND= <value>, what should be this value? 915E6 or 8/9 according to the <MKRWAN.h>?

sabas1080, 2018-07-27:
it can be for the subband, although I still do not know how to configure that

conradholt, 2018-07-27:
I feel they should provide more information on this matter, because there's two channel plans on the US915 (referring to upstream only) one with 8 channels using Lora's 500Khz bandwidth and 64 channels using Lora's 125Khz bandwidth... 
Also provide documentation for the AT commands.
I hope they fix this issue with the US LoRa connection soon...

akash73, 2018-07-30:
Hi, 
pls look at this 
https://www.arduino.cc/en/Tutorial/MKRWANFirstConfiguration

inside the example : 

```
Serial.println("Register to your favourite LoRa network and we are ready to go!");
  // change this to your regional band (eg. US915, AS923, ...)
  if (!modem.begin(EU868)) 
```

change the value of (!modem.begin(EU868))  to US915

check if works.
thanks

conradholt, 2018-07-31:
Hi akash73,
I've tried the MKRWANFirstConfiguration example but still no success... I also wanted to ask about the code on the gateway, currently I'm using a Raspberry Pi 3b with a dragino shield wired I'm using the single_channel_pkt_fwd from https://github.com/tftelkamp/single_chan_pkt_fwd. My question is; is the Arduino MKR doing frequency hops between 8 or 64 channels? Is the code from the single channel packet fwd also doing frequency hops (probably only 8 channels)? Any suggestion on which single_chan_pkt_fwd use?

Kind regards:)

conradholt, 2018-08-09:
Any update? Have you been able to securely transmit data (OTAA,ABP) using LoRa standards over the 915MHz?

sslupsky, 2018-09-07:
I just fired one up today on US915 and I have been able to send packets through my gateway to my server.  There are some issues to resolve but the packets are getting there.

conradholt, 2018-09-12:
Hey @sslupsky could you specify the setup you used? Gateway, if possible post the gateway code, and I'm assuming you tried the mkr with the MKRWANFirstConfiguration example?
Thanks

sslupsky, 2018-09-12:
I am using the Laird gateway.  I am not sure what you mean by gateway code?  When using the MKRWAN1300 with an 8 channel gateway such as the Laird, select the "US915_Hybrid" region.  That appears to work most reliably to join the network.  I have configured my Laird gateway for channels 0 - 7 and 64.  I have not actually figured out how the "Hybrid" configuration differs from the standard one yet though.  It is not documented anywhere that I can find.

I've used the FirstConfiguration example and the LoraSendandReceive example and gone from there.  Though, I have not actually tested receiving yet.  There is some weirdness regarding resetting the board.  If the board is reset by pressing the reset button or uploading a new firmware, the setup() runs, it joins the network.  The loop() runs once and sends one uplink and hangs.  If I do a power on reset, the loop() function cycles as it should.

A little bit off topic, I am having a serious problem with power consumption though.  I have not figured out how the get the sleep current to less than 1.16mA.  I have configured my unused inputs to use internal pull-ups and achieved a sleep current of 1.16mA which is two orders of magnitude to high.  I did dig into the Arduino libraries and it appears the clock domains are configured to use the XOSC32K oscillator for the RTC and the USB serial port is being shut down prior to sleep, so I think it should be possible to achieve a sleep current of 10uA or less.  I've been searching around the web regarding this issue and no one seems to have an answer though the question has been asked a few times.  I am surprised no one from Arduino has addressed this issue yet.  IMHO it makes the module unusable for applications that are required to operate from a battery.  At the moment I am not sure if the issue can be resolved by updating the firmware or if its a hardware problem.  I am using the latest 1.1.6 firmware.

conradholt, 2018-09-22:
thanks @sslupsky for the reply!.
I have been able to set an arduino uno with a dragino shield v1.4 to send data to ttn using a single channel gateway (sx12xx) like a Raspberry pi + dragino shield or the dragino lg01. 
Since the gateway it's a single channel, when sending data I have to make sure I disable the 63 other channels for the us band (just sending data on 902.3Mhz) ... Is there any way to disable all the channels except 1 for mkr1300? 
Also regardless of OTAA or ABP do you have to give the APPKey or APPSKey always in MSB format because that's the case for the arduino uno when connecting to ttn through abp.
cheers!

conradholt, 2018-09-24:
Thanks @sslupsky I will definitely do that soon, but as of now I need to work with the single channel gw!

conradholt, 2018-10-10:
so @sslupsky I've been able to setup a lorawan network with my own private server using mkr's and uno's and a kona pico gw from tektelic. The reason why the US915_HYBRID works better than the US915 has to do with the amount of channels you have in your gateway, in my case 8. That's what the US915_HYBRID is doing, is hoping between 902.3-903.9.
I am able to see the packets with the payload in my server, but everytime I send a packet the serial (for the mkr) says "error sending the message", and still the packet is perfectly received at the gw and pushed to the server, so i guess it has to do with either the downlink or some sort of ack.
Any thoughts?

sslupsky, 2018-10-22:
Are you using the modem.endpacket() function?  You get that message if the ack is not received or if the payload is more than 64 bytes.  I noticed that I get the occasional error.  For testing, I send a packet every 10 sec.  It appears about every 6 min I see a packet failure.  I noticed that the radio behaves odd when this happens.  I am wondering if it is doing a rejoin.