arduino-libraries/MKRWAN #28: The sleep() method is not implemented

AmedeeBulle, 2018-09-04:
The [sleep()](https://github.com/arduino-libraries/MKRWAN/blob/master/src/MKRWAN.h#L585) method is not implemented  (There is no `AT+SLEEP` command defined in the firmware)...

facchinm, 2018-09-04:
Sleep is automatically handled by the firmware running on the module (https://github.com/arduino/mkrwan1300-fw) , the API is related to Murata firmware. I'll add this information to the sources and return `true` if the running fw version is not Murata's.

AmedeeBulle, 2018-09-04:
Thank you.

To clarify, what you are saying is that the module will sleep by itself when idle, and there is no need to call sleep() for low power operation?

facchinm, 2018-09-04:
Exactly :slightly_smiling_face:

sslupsky, 2018-09-13:
Has anyone confirmed the sleep current of the current Murata firmware?  I am having difficulty getting the sleep current of the MKRWAN1300 below 1.16mA.  I believe the SAMD21 is properly sleeping and my unused pins are configured as inputs with pull ups.  I am beginning to think the excessive sleep current is due to the Murata module.

closed 2018-09-20 by facchinm