arduino-libraries/MKRWAN #29: Port number changes

sslupsky, 2018-09-07:
I am not sure if this is a bug yet.

While experimenting with the example code, I noticed is that the port number changes from time to time.  I have not figured out why it changes.  I have seen in change when you reset the device.  I have also seen what I think is just a random change.  That is, several hours had past since the last uplink and I sent another uplink and the port number changed on the new uplink.

I've seen uplinks using port number 2, 4 and 6.  Looking further into this, I cannot find the API for setting the port number for an uplink.

sslupsky, 2018-09-09:
I found an API call to set the port in MKRWAN.h

bool setPort(uint8_t port)

I am still unsure still why the port number is changing.  At the moment I do not specifically set the port number at any point in the code so I am wondering what method the MKRWAN firmware uses to set the port number?

sslupsky, 2018-09-16:
This appears to be a mkrwan-fw issue.

https://github.com/arduino-libraries/MKRWAN/issues/8

sslupsky, 2018-09-18:
I am re-opening this issue.  I programmed a new MKRWAN1300 today and when the sketch makes a call to setPort(10), the port number does not get set correctly.  The server is showing that the messages are being sent on Port 21.  I power cycled the board and now the messages are sent on Port 19.  I reprogrammed the board and now the messages are being sent on Port 20.  

In all cases the messages should be sent on Port 10.

facchinm, 2018-09-19:
Which firmware version are you running? Could you try reflashing the FW by running MKWANFWUpdate_standalone sketch and then trying again?

sslupsky, 2018-09-19:
Hi @facchinm Oops, yeah, that was it.  When I started testing the second board I forgot to update the firmware.  Thank you.

closed 2018-09-19 by sslupsky