arduino-libraries/MKRWAN #32: mkrwan with single channel gw

conradholt, 2018-09-24:
So, after trying to set up (without success) the mkrwan with a single channel gateway I am asking if someone has been able to do that? 
Currently the mkr fw doesn't have the option (like the uno's) to disable channels, I would like the mkr to send data just on the 902.3 Mhz channel not the rest.

facchinm, 2018-09-25:
Hi @conradholt ,
what do you mean with "like the UNO"? The Semtech chip inside the Murata module can be controlled independently (using this library https://github.com/sandeepmistry/arduino-LoRa ) if you want to implement a point-to-point communication (which I believe is what a 1channel "gateway" does.

conradholt, 2018-09-25:
Sorry for the misunderstanding @facchinm , I hope this makes things a bit more clear,

One of the nodes is an arduino uno + dragino v1.4 shield using the lmic library (https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/ttn-abp/) with a small modification in the set-up:

LMIC_setDrTxpow(DR_SF7,14);
    for (int i=1;i<64;i++)
    {
      LMIC_disableChannel(i);  // only the first channel 902.3Mhz works now. 
    }

The gateway is the dragino LG01 using the sx1276 semtech transceiver running the single_channel_pkt_fwd (https://github.com/tftelkamp/single_chan_pkt_fwd).
This set-up is able to send data to ttn, which i'm also trying to accomplish using the arduino mkr1300 as another end-node. I am positive that because I can't set the mkr to only send data on the 902.3Mhz my gateway is not receiving any packet from the mkr (maybe if i leave it overnight the gw will detect some packets but with a lot of them being missed). I am not trying to set a point-to-point communication, I am trying to set a LoRaWAN network using a single_channel gateway and multiple nodes.

facchinm, 2018-09-26:
@conradholt you can still use `lmic` with the MKRWAN in this way:
before calling `os_init` you need to instantiate a LoRaModem and call `dumb()` on it.
From then on, you can use SPI1 and LORA_IRQ_DUMB pins and run all the stack in the D21 itself https://github.com/arduino-libraries/MKRWAN/blob/master/src/MKRWAN.h#L541
The pinmapping should become 
```
const lmic_pinmap lmic_pins = {
    .nss = LORA_IRQ_DUMB,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LORA_RESET,
    .dio = {LMIC_UNUSED_PIN, LMIC_UNUSED_PIN, LMIC_UNUSED_PIN},
};
```
and you should change `SPI` into `SPI1` [here](https://github.com/matthijskooijman/arduino-lmic/blob/ba1265d5d2f775177cdc7c82186724e4f0bdc3a8/src/hal/hal.cpp#L77
)

Let me know if it works for you!

conradholt, 2018-09-26:
@facchinm I haven't been able to make it work, let me walk through the steps to see if I've missed something.
using the example from the abp library https://github.com/matthijskooijman/arduino-lmic/blob/ba1265d5d2f775177cdc7c82186724e4f0bdc3a8/examples/ttn-abp/ttn-abp.ino I have included the mkrwan library, then I've instantiated (as global) a LoRaModem modem, and I've called modem.dumb() in the setup before os_init. Then I've created the pin mapping as you mentioned. Next I've changed SPI to SPI1 (but not SPI_MODE0 this still the same right?) in the hal.cpp file and uploaded this new lmic.h library.
At first I'm getting errors from the ASSERT functions in hal.cpp (FAILURE /tmp/505991622/custom/arduino-lmic/src/hal/hal.cpp:23) if I comment this ASSERT then line :24 complains, if I comment this line then radio.cpp:689 complains, if I comment then radio.cpp:545, finally if I comment this then there's no error on the serial monitor but the code is "frozen" somewhere because I'm not able to see the prints from my main code...

There might be some problem with the pin mapping? because the ASSERTS basically check:
ASSERT(lmic_pins.dio[0] != LMIC_UNUSED_PIN);
ASSERT(lmic_pins.dio[1] != LMIC_UNUSED_PIN || lmic_pins.dio[2] != LMIC_UNUSED_PIN); 

Any thoughts on what to do next?

facchinm, 2018-09-27:
Ah, my fault, lmic requires DIOs to serve interrupts (https://github.com/matthijskooijman/arduino-lmic/blob/ba1265d5d2f775177cdc7c82186724e4f0bdc3a8/src/hal/hal.cpp#L69) :slightly_frowning_face: We don't have enough exposed pins to handle this; there could be other libraries that do not need the IRQ pins but I'm not aware of them

ivanpiffer, 2019-01-23:
so @conradholt tell me ... did you resolve it? I'm in the same situation ... the dragino LG01_P and I don't know how to set up my mkr wan 1300 with only 1 channel :(

conradholt, 2019-01-31:
unfortunately @ivanpiffer the sam21d or the mkr series won't support the single channel configuration because it's doing frequency hops on more than one frequency. If you get an arduino uno or any arduino working on AVR then the LMIC example code works just fine.

Finally I purchased a multichannel gateway (LAIRD sentrius) and the mkr was plug and play for both OTAA/ABP.

conradholt, 2019-01-31:
@facchinm you can close this issue since the mkr won't support single channel

facchinm, 2019-02-01:
Ok, I'm sorry it didn't end well. Closing as wontfix

closed 2019-02-01 by facchinm

deanpretorius, 2019-07-29:
I am stuck in the same situation with a bunch of these boards and a single channel gateway. Is there a way to fix this? 
> Ah, my fault, lmic requires DIOs to serve interrupts (https://github.com/matthijskooijman/arduino-lmic/blob/ba1265d5d2f775177cdc7c82186724e4f0bdc3a8/src/hal/hal.cpp#L69) 🙁 We don't have enough exposed pins to handle this; there could be other libraries that do not need the IRQ pins but I'm not aware of them

Are there any other, more recent, libraries that do not need IRQ pins, or is there a way of making pins available to handle these interrupts by disabling unneeded services on the MKR WAN?

facchinm, 2019-07-30:
Hi @deanpretorius ,
single channels gateways are not compatible with LoRaWAN at all since they don't implement channel hopping. The cheapest multichannel gw (https://www.thethingsnetwork.org/docs/gateways/thethingsindoor/) works just fine. https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/pull/57 allows you to program the LoRa module directly, so with a bit of effort the same result can be achieved but there's nothing ready to use yet.