arduino-libraries/MKRWAN #33: Imposssible to receive 0x00 in the received message payload

ruipalma, 2018-10-15:
I can confirm the behaviour reported by vahisa in https://forum.arduino.cc/index.php?topic=536628.0, 
The modem.read() method do not work with 0x00 payload.

facchinm, 2018-10-16:
Will take a look, thanks for reporting

ruipalma, 2018-10-16:
Using the [TtnModemTest.ino](https://github.com/AmedeeBulle/TtnModemTest) to interact directly with the modem I can see that the modem is not transmiting the 0x00 bytes after a +RECV. So the problem seems to be in the murata firmware.

ruipalma, 2018-10-16:
The bug is in the murata firmware.
The command AT+FORMAT only apply to the sended messages,no the received ones.
The received messages are processed in the function

`ATEerror_t at_Receive(const char *param)
{
  AT_PRINTF("+RECV=");
  AT_PRINTF("%d,%d\r\n\n", ReceivedDataPort, ReceivedDataSize);
  if (ReceivedDataSize)
  {
    AT_PRINTF("%s", ReceivedData);
    ReceivedDataSize = 0;
  }
  AT_PRINTF("\r");

  return AT_OK;
}`

As you can see the received data are scanned as a string, so the 0x00 is a terminator.
Maybe we can use the "format_send_v2" to process the received data also.
Something like:

`ATEerror_t at_Receive(const char *param)
{
  AT_PRINTF("+RECV=");
  AT_PRINTF("%d,%d\r\n\n", ReceivedDataPort, ReceivedDataSize);
  if (ReceivedDataSize)
  {
    if (format_send_v2==0)
       AT_PRINTF("%s", ReceivedData);
    }
    else
    {
      for (unsigned i = 0; i < ReceivedDataSize; i++)
      {
        AT_PRINTF("%02x", ReceivedData[i]);
      }
    }
  }
  ReceivedDataSize = 0;
  AT_PRINTF("\r");
  return AT_OK;
}`

ruipalma, 2018-10-16:
OK. I have done the modification from the last post in the murata fw and I can confirm that its working. The modem sends a HEX string when +DFORMAT=1.
I attached the modified fw. I keep the version.
[fw.zip](https://github.com/arduino-libraries/MKRWAN/files/2483302/fw.zip)

ruipalma, 2018-10-16:
Hi @facchinm, I have done the PR.

facchinm, 2018-10-23:
Fixed by merging https://github.com/arduino/mkrwan1300-fw/commit/e547242a651568b7fefe0d5b543a6ff6022619f5

closed 2018-10-23 by facchinm