arduino-libraries/MKRWAN #34: Firmware update does not compile

sslupsky, 2018-10-27:
I pulled the 1.1.8 version today and attempted to upgrade the firmware.  I received the following error during the verify (compile) phase:

`Verifying...
/Users/stevenslupsky/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: MKRWANFWUpdate_standalone.ino.elf section `.data' will not fit in region `RAM'
/Users/stevenslupsky/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
/Users/stevenslupsky/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 46632 bytes
collect2: error: ld returned 1 exit status
exit status 1
[Error] Exit with code=1`

facchinm, 2018-10-29:
Hi @sslupsky , my fault, I didn't push the last commit, doing it now with https://github.com/arduino-libraries/MKRWAN/commit/f0238b873bbbc86dd2949729ae687bb2910e21c4 . Thanks for pointing it out!

closed 2018-10-29 by facchinm

sslupsky, 2018-10-30:
That fixed it.  Thank you!  👍