arduino-libraries/MKRWAN #38: modem.restart() hangs

sslupsky, 2018-11-07:
The modem.restart() function does not return.  It appears to hang on:

    sendAT(GF("+REBOOT"));

sslupsky, 2018-11-09:
Ok.  I see the problem now.  I called modem.restart() before modem.begin() to force a reset of the radio when the MKRWAN boots.  This is in case there is some form of reset other than a POR.  That is, if the radio is the problem, then reboot it.  I did not realize modem.begin() needed to be called before this function.

So, I think if you modify your example you will experience the hang.

facchinm, 2018-11-12:
`begin()` already resets the module, so `restart()` is only needed for "runtime restart". The band selection must be reapplied through another `begin` (or `configureBand`) though; I can understand the problem, will try to think about a proper API that leaves no space for ordering issues.

sslupsky, 2018-11-13:
Just to clarify, I do not recall that begin() resets the module, rather it resets the modem object?

facchinm, 2018-11-13:
`begin()` toggles RESET pin (https://github.com/arduino-libraries/MKRWAN/blob/master/src/MKRWAN.h#L406) , so it reboots the module, in a certain way :slightly_smiling_face:

sslupsky, 2018-11-13:
Oh, for some reason I though that was excluded by the #ifdef SerialloRa?  The default condition is that SerialloRa is defined?  I could not find it defined anywhere and I do not do so in my code.

facchinm, 2018-11-13:
It's defined in MKRWAN core, so it can adapt to any other board using the Murata module.

sslupsky, 2018-11-13:
Ahh, that sheds a whole new light on my understanding of things.  Thank you for that clarification.  So, SerialloRa is intended to be defined when you use the SPI interface to communicate the the Murata module?  (For some reason I misinterpreted its' intended use for when the UART interface is used).

sslupsky, 2018-11-14:
Since begin() does a module reset calling restart() then begin() is redundant.

I suppose one could call restart() then configureBand() to avoid a pin reset of the module.  Is there any reason to avoid a pin reset?  If not, should restart() should be removed from the API?  

You may want to document on the Arduino website that begin() does a pin (hardware) reset of the module.