arduino-libraries/MKRWAN #42: Can't send data when using RTCzero

Oretoh, 2018-12-18:
I have a really weird problem with the Arduino MKR 1300 and the use of the RTCzero library. Simply put when I use both of them the data does not get sent and I have no idea why. The alarms when not trying to send data work normally and the modem.print(msg) also works normally when I am not using the RTC library.

The code looks like this:

```
    #include <EmonLib.h>
    #include <RTCZero.h>
    #include <MKRWAN.h>
    
    
    EnergyMonitor emon1;
    EnergyMonitor emon2;
    EnergyMonitor emon3;
    RTCZero rtc;
    LoRaModem modem;
    
    String appEui = "1234567891011121";
    String appKey = "ffffffffffffffffffffffffffffffff";
    
    /* INITIAL_TIME */
    const byte seconds = 0;
    const byte minutes = 0;
    const byte hours = 0;
    const byte day = 17;
    const byte month = 12;
    const byte year = 18;
    
    
    byte second_alarm = 0;
    byte minute_alarm = 0;
    byte hour_alarm = 0;
    byte INTERVAL = 60;
    int SEND_LOOP = 2;
    
    int totalKW;
    int time_running = 0;
    
    void setup()
    {
      Serial.begin(115200);
      if (!modem.begin(EU868)) {
        Serial.println("Failed to start module");
        while (1) {}
      };
      Serial.print("Your module version is: ");
      Serial.println(modem.version());
      Serial.print("Your device EUI is: ");
      Serial.println(modem.deviceEUI());
    
      Serial.println("Connecting");
      int connected = modem.joinOTAA(appEui, appKey);
      if (!connected) {
        Serial.println("Something went wrong; are you indoor? Move near a window and retry");
        while (1) {}
      }
      Serial.println("Connected");
    
      // Set poll interval to 60 secs.
      modem.minPollInterval(60);
    
      analogReadResolution(9);
      emon1.current(1, 53);
      emon2.current(2, 53);
      emon3.current(3, 53);
      time_running = 0;
    
      rtc.begin(); // initialize RTC
    
      rtc.setAlarmTime(hour_alarm, minute_alarm, second_alarm);
      rtc.enableAlarm(rtc.MATCH_HHMMSS);
      rtc.attachInterrupt(ISR);
    
      // Set the time
      rtc.setHours(hours);
      rtc.setMinutes(minutes);
      rtc.setSeconds(seconds);
    
      // Set the date
      rtc.setDay(day);
      rtc.setMonth(month);
      rtc.setYear(year);
    
    
    }
    
    void loop()
    {
    }
    
    void ISR() {
    
      int totalWatt = 0;
      unsigned long delay_send = 0;
      int sending = 0;
    
      double Irms1 = emon1.calcIrms(600);
      if (Irms1 < 0.3) Irms1 = 0;
      double Watt1 = Irms1 * 230;
    
      double Irms2 = emon2.calcIrms(600);
      if (Irms2 < 0.3) Irms2 = 0;
      double Watt2 = Irms2 * 230;
    
      double Irms3 = emon3.calcIrms(600);
      if (Irms3 < 0.3) Irms3 = 0;
      double Watt3 = Irms3 * 230;
    
      totalWatt = Watt1 + Watt2 + Watt3;
      totalKW = totalKW + totalWatt / 1000;
      Serial.println(time_running);
    
      sendDataChecker();
      setAlarm();
      time_running = time_running + 1;
    }
    
    void sendDataChecker() {
      if (time_running == SEND_LOOP) { 
        double IrmsTotal = Irms1 + Irms2 + Irms3;
        String msg = "{\"id\":\"avac_aud1\",\"kW\":" + String(totalKW) + ", \"current\":" + String(IrmsTotal) + "}";
        int err;
        Serial.println("Ready to Send");
        modem.beginPacket();
        modem.print(msg);
        err = modem.endPacket(true);
        Serial.println("Sent1");
        if (err > 0) {
          //message sent correctly
          Serial.println("Sent");
          time_running = 0;
          totalKW = 0;
        } else {
          Serial.println("ERR");
          time_running = 0;
        }
      }
    }
    
    void setAlarm() {
      second_alarm += INTERVAL;
      if (second_alarm >= 60) {
        minute_alarm++;
        second_alarm = 0;
      }
      if (minute_alarm >= 60) {
        hour_alarm++;
        minute_alarm = 0;
      }
      if (hour_alarm >= 24) {
        hour_alarm = 0;
      }
    
      rtc.setAlarmTime(hour_alarm, minute_alarm, second_alarm);
    }
```

And here's the output:

```
    Connected
    0
    1
    18:13:33.158 -> Ready to Send
```

Doesn't go beyond that and the data doesn't get sent. I realize this isn't really stack overflow or the best place to ask these kind of questions but I can't find an answer anywhere else. The code seems fine, it's just that the data won't send when using RTC.

sslupsky, 2018-12-18:
I do not think you want to call the modem functions beginPacket() and endPacket() in your ISR.  Depending on the state of the LoRa stack, they can take a *long* time to execute.

Oretoh, 2018-12-18:
I managed to get it working, the problem lies in the packet sending code not being inside the loop method. Once I added the ISR method to the loop() it worked normally so I just made the alarm interrupt trigger a boolean every minute and ISR is inside an if that verifies if the boolean is true or not.

Sure it's not the most elegant solution and it keeps the arduino in an active loop but it gets the job done.

facchinm, 2019-09-19:
Closing as fixed based on previous comment

closed 2019-09-19 by facchinm