arduino-libraries/MKRWAN #43: Receive more than 64 bytes

kraef, 2018-12-20:
Is it possible with the mkr wan to send and receive a message with size more than 64 bytes??

tohliheng, 2020-04-04:
I have been using the **RAK811 shield** for Arduino for 2 months. On that shield, **I was not able to send a payload of larger than 64 bytes even at a DR larger than 3.**

[image]
This table shows the size limit of payload when the module is set to different DR. (Source: [LoRaWAN™ 1.0.2 Regional Parameters](https://lora-alliance.org/sites/default/files/2018-05/lorawan_regional_parameters_v1.0.2_final_1944_1.pdf).)

I am curious if the MKR WAN 1310 has the same limitation as well before I get one. Thank you.

flhofer, 2021-05-02:
Unfortunately it's hardcoded in the middleware and chip driver for the 1.2.x firmware. Maybe that changes for the newer releases

flhofer, 2021-07-08:
I performed a [update](https://github.com/arduino/mkrwan1300-fw/pull/35) to the firmware which now supports 242 bytes HEX or 121 bytes binary send, and 242 bytes receive (= 250 incl Mac header). Local tests seem ok. Should be merged soon to fw ver 1.2.4.