arduino-libraries/MKRWAN #45: Problem when updating module murata

sabas1080, 2019-01-18:
I have 20 arduino MKR1300 

I am updating 3 arduino mkr1300 with example of "MKRWANFWUpdate_standalone"

https://github.com/arduino-libraries/MKRWAN/tree/master/examples/MKRWANFWUpdate_standalone

but update does not finish in 3 arduino MKR1300

[image]

facchinm, 2019-01-23:
Hi @sabas1080 ,
sorry for the late response.

The 3 boards that are not updating are working fine with standard firmware (`FirstConfiguration` example) ? If so, it may be a timing issue, since the factory flash is executed using a slightly modified version of `MKRWANFWUpdate_standalone` sketch so any sold board should be able to update the fw.

Let me know! Thanks

sabas1080, 2019-01-23:
in one FirstConfiguration example not work

in all devices the monitor serial break and not respond

sabas1080, 2019-01-24:
in other boards not break, I have test in other computer with windows same error

pelikhan, 2019-02-02:
Just got 4 boards from Amazon, 3 of them are bricked after trying the update script. (connect with Serial Monitor, try to press keys, nothing happening). The boards won't come back to life.

facchinm, 2019-02-04:
@pelikhan it's physically impossible to brick a board using the updater sketch (since its only writes the Murata module memory). What happens if you double click the reset button? The orange LED should start fading and the board should be in bootloader mode again.

sabas1080, 2019-02-18:
I am buying more mkr 1300 boards and this is becoming a very common error @facchinm

facchinm, 2019-02-18:
It might be related with a module variability at this point; just to make sure it's not IDE related, would you mind (both @pelikhan and @sabas1080 ) to post your PC specs, SAMD core version, MKRWAN library version (attach it here if you patched it) so I can try reproducing locally with a board from a recent batch? Thanks

pelikhan, 2019-02-18:
How can we get that info?

pelikhan, 2019-02-18:
Is there any diagnostic Arduino sketch we can run?

sslupsky, 2019-02-18:
Could these random update issues be related to the power supply issue with the Murata module?  That is, the module on unmodified boards has an incorrect voltage applied when using the USB port.

sabas1080, 2019-02-18:
Arduino IDE 1.8.8 and 1.8.6
MKRWAN version: 1.0.9
OS: MAC OS High Sierra
SAMD Core: 1.6.20

pelikhan, 2019-02-18:
I don’t have the board at hand but my issues were on Windows 10.

per1234, 2019-02-18:
>How can we get that info?

@pelikhan Here's how to find the SAMD core version:

1. Tools > Board > Boards Manager
2. Wait for downloads to finish
3. Scroll down until you see "Arduino SAMD Boards (32-bits ARM Cortex-M0+) by Arduino version x.y.z INSTALLED", where x.y.z is the version you have installed.
1. Click "Close".

---
Here's how to get the MKRWAN library version:

1. Sketch > Include Library > Manage Libraries
2. Wait for download to finish
3. In the "Filter your search" field, type "MKRWAN". The search result should show "MKRWAN by Arduino version x.y.z INSTALLED", where x.y.z is the version of the library you have installed.

facchinm, 2019-02-19:
@sabas1080 would you mind sending me the serial number of the malfunctioning boards? To obtain it, connect a board and select "Tools/Get Board Info" from the IDE. Any sketch must be running (no bootloader mode, blink is enough).
Thanks a lot

facchinm, 2019-02-19:
@sabas1080 @pelikhan would you mind testing this patch https://github.com/arduino-libraries/MKRWAN/commit/d994574bedf73c970a41a3d3b11d439e33f49d8c ? I was able to reproduce the hang at SAMD level and to fix it by setting `cmd_get_reply` variable to NULL.
On my boards the Murata always responded correctly anyway, so I hope it's only a sw issue.

sabas1080, 2019-02-19:
@facchinm fix problem! 4 boards :D

closed 2019-02-19 by sabas1080

pelikhan, 2019-02-20:
How do I deploy this patch?

sabas1080, 2019-02-20:
@pelikhan I modified the sketch directly, file .h only add "=null"