arduino-libraries/MKRWAN #46: Setting the transmit power

lodesmets, 2019-02-21:
This function isn't documented: `bool power(_rf_mode mode, uint8_t index)` and the arguments aren't named very good. The mode argument I can understand, but what is index? Is it the transmit power? Between what values can it change?

facchinm, 2019-02-21:
Hi @lodesmets ,
that function signature is a leftover of the Murata firmware compatibility layer (which I belive we can abandon sooner or later). It maps in the firmware to
https://github.com/arduino/mkrwan1300-fw/blob/63787fe5ed8bd07119caba20d2065a26004b2261/Projects/Multi/Applications/LoRa/AT_Slave/src/at.c#L382-L397
So `mode` is ignored while `index` is the transmit power (from 0 to 5)
https://github.com/arduino/mkrwan1300-fw/blob/188a5cd9b595d342b6c00d3dc9e9b9f7a102ef6b/Projects/Multi/Applications/LoRa/AT_Slave/src/command.c#L217
Any proposal on how to improve the function name or add some documentation is well accepted :wink:

lodesmets, 2019-02-21:
> Any proposal on how to improve the function name or add some documentation is well accepted 😉

Maybe call it transmitPower instead of index that would be a more explaining variable name

facchinm, 2019-02-21:
Would you mind submitting a pull request? Thanks!

lodesmets, 2019-02-21:
> Would you mind submitting a pull request? Thanks!

Done, I just made the argument have a more sensible name. But I don't know what the stronger transmit power is (I guess 5)

lodesmets, 2019-02-21:
Pressed the wrong button,
I know it is a lot of work, but why not document all the functions?

facchinm, 2019-02-21:
> why not document all the functions?

Because it's a lot of work :smile: 
Jokes apart, this library was born to handle the Murata firmware; when we decided to switch to the open firmware module we thought it was a good idea to keep the library compatible.

It was not a great idea in the end, so the long term target is to rewrite the firmware entirely and expose the functionalities in a saner way (and, last but not least, document them properly)

facchinm, 2019-02-21:
Ah, I merged the PR, so I'm closing the issue

closed 2019-02-21 by facchinm

ElectronicallyE, 2021-05-24:
Been playing around with the command `modem.power(PABOOST, X)`;

Based upon [this](https://forum.arduino.cc/t/transmit-power/575415/2) Arduino Forum post, 0 is the highest TX power.

I've done some testing on 0 and 5 and there doesn't seem to be a noticeable difference in SNR or RSSI, no matter if the gateway is close or far from the sensor. One transmission, 0 has a better RSSI and SDR, on another it's 5. So either, the command isn't working or the difference is negligible.

I'm setting up MKR WAN 1310 and the transmission as follows:
```
modem.setADR(false);
modem.dataRate(0);
modem.beginPacket();
modem.print("AA");
modem.endPacket(true);
```

Any ideas why I'm not seeing a difference?