arduino-libraries/MKRWAN #49: Downlink message containing 00 does not work

miqmago, 2019-03-02:
I'm having exactly this problem described here: http://forum.arduino.cc/index.php?topic=536628.0

Can't send any payload containing `00` byte inside. The device gets stuck. If the payload does not contain any `00` then all works extremly fine.

Don't know if already reported here, if it is a problem of MKRWAN library or modem problem...

Board: MKR WAN 1300

facchinm, 2019-03-04:
Hi @miqmago ,
did you update the firmware to the latest version (1.1.9) ? Try running `MKRWANFWUpdate_standalone` example and report back if it fixes the issue (see https://github.com/arduino-libraries/MKRWAN/issues/33)

miqmago, 2019-03-05:
Hi @facchinm, thanks. Worked great.
As it was difficult to find a step by step firmware upgrade, I leave here the only way it worked for me (tried several ways): http://tinkerman.cat/arduino-mkr-wan-1300/

closed 2019-03-05 by miqmago