arduino-libraries/MKRWAN #51: Firmware code/project for LoRa module should be made available and/or referenced in docs

apullin, 2019-03-17:
It appears that there is no documentation on or about the firmware that goes onto the STM32L0 inside the Murata module.
The binary blob is in the standalone-updater example, but not the source.

Including the complete source inside this library would likely not be the right place for it.
But it should still be made public somewhere and linked to in the documentation.

If the firmware is *only* the AT_slave example from the STM32+SemTech stack LoRa software package, then at least that should be documented, for completeness and clarity.
But it is unclear if any patches are added on top of that distribution.

sabas1080, 2019-03-18:
The code is available in https://github.com/arduino/mkrwan1300-fw

facchinm, 2019-03-18:
I'm adding some lines on README, thanks for pointing that out.

closed 2019-03-18 by facchinm