arduino-libraries/MKRWAN #52: Downlink not work - OTAA

IManux, 2019-03-18:
Hi! I'm making test with the arduino mkr 1300  + TTN, and the downlink not work, the ttn application data by node show me than the payload bytes was sent (downlink) but in the monitor serial (arduino) not receive nothing... the uplink works I'm sending data every 5 minutes.

Note1: I have upgrade to firmware 1.1.9 with the example in the repository: MKRWANFWUpdate_standalone.
Note2: The downlink payload are bytes and I am sending 5 bytes.
Note3: The uplink are 5 bytes and I'm using joinOTAA.
Note4: I am setting the port, I'm using port 2.
Note5: I have tested with dateRate(0) and dateRate(5).

I hope to can somebody help me.
Regards

facchinm, 2019-03-18:
Hi @IManux ,
are you using _bytes_ or _fields_ as payload format?

IManux, 2019-03-18:
Hi @facchinm 
I'm using bytes as payload format.
[image]
In the image you can to see, the ttn indicate then downlink was sent, but the arduino not receive nothing to display in monitor serial.

facchinm, 2019-03-18:
Bytes downlink should work just fine, I've always tested them with TTN and never had a problem. Does the LoraSendAndReceive example work or are you getting the same problem there too?

IManux, 2019-03-18:
This is my sketch:
[SketchNode002.txt](https://github.com/arduino-libraries/MKRWAN/files/2978447/SketchNode002.txt)
I use the example: "LoraSendAndReceive" like reference. I am sure my sketch is not wrong. maybe can you test it to verify the problem is not the sketch.

IManux, 2019-03-18:
The most surprising thing is that I have two boards arduino mkr 1300 (one board bought from arduino.cc) and I get the same problem.... the downlink not work.... other thing I have noted... when I send the data (uplink) the modem.endPacket(true) always give me 'false' then the monitor show "error" (this mean data not sent) but it's false bez the data is received in the ttn.

facchinm, 2019-03-21:
I just tested your sketch with this setup:
- mkrwan with 1.1.9 fw
- TTN configured as OTAA
- very near gateway (~10meters) so I removed the antenna from the mkrwan to make it more realistic.

And everything works perfectly, both upstream and downstream. Which gateway are you using? Have you tried transmitting and receiving data with another non mkrwan board?

IManux, 2019-03-23:
excuse me the delay facchinm, I had making more tests and I found the issue.
when I started to make test with the arduino board mkr1300 I never get connect me to network I tried in two location in the roof (5th floor) and nothing.... (and the gateway more close to 1KM), then I setup an small fipy gateway nano and I started to get success connection to network and send uplink success but not downlink(with the arduino mkr), It's strange... it's like the fipy gateway nano help at the arduino board to get success connection at the network.
Well Then I follow making test and I decide to test the arduino mkr (alone) in other location (other building roof) and I get success connection, uplink and downlink, it's strange I have tested in 3 location very close to gateway public (1KM) and only one location is success.

IManux, 2019-03-23:
excuse me the delay facchinm, I had making more tests and I found the issue.
when I started to make test with the arduino board mkr1300 I never get connect me to network I tried in two location in the roof (5th floor) and nothing.... (and the gateway more close to 1KM), then I setup an small fipy gateway nano and I started to get success connection to network and send uplink success but not downlink(with the arduino mkr), It's strange... it's like the fipy gateway nano help at the arduino board to get success connection at the network.
Well Then I follow making test and I decide to test the arduino mkr (alone) in other location (other building roof) and I get success connection, uplink and downlink, it's strange I have tested in 3 location (in the roof) very close to gateway public (1KM) and only one location is success.

InzynierDomu, 2019-05-23:
Hi @facchinm , I have the same problem.
ABP, Arduino SAMD Board ver: 1.6.21, ver lib: 1.0.10
Her is my code, writ with the example: https://github.com/InzynierDomu/ArduinoMKR1300/blob/master/ArduinoMKR1300.ino
After many downlink one received.
Screen from TTN: https://github.com/InzynierDomu/ArduinoMKR1300/blob/master/lorawan_arduino_err.PNG
Log: https://github.com/InzynierDomu/ArduinoMKR1300/blob/master/log.txt
In log in line 9044 
“Received: 12 1” Weird, because is received after 110 seconds after uplink. In class A it’s can't be possible. On TTN I see a problem with counter, iterated after few msg, not every.
In ABP Uplink work fine but end packet return -1. OTAA sended join and not connected.
What can I try to test?

aalbinati, 2021-01-03:
@facchinm 
I am having the same problem with my Arduino MKR 1310. Even if the gateway is really close most packets are not heard by the node.
What could be happening?