arduino-libraries/MKRWAN #55: 32 bits frame counters

giampiero7, 2019-04-30:
`setFCU()` and `setFCD` accept 16bits values.
Why are 32bits values not supported?
Thanks!

flhofer, 2021-07-08:
The counters are the ones defined by the LoRa standard which are 16 bit numbers used for frame identification. See `4.3.1 Frame header (FHDR)` of the LoRaWan standard 1.0.3

giampiero7, 2021-07-09:
Hi,
Old topic, but still relevant. Refer to `4.3.1.5 Frame counter (FCnt)`:

> ...
> The LoRaWAN allows the use of either 16-bits or 32-bits frame counters.
> ...

flhofer, 2021-07-10:
@giampiero7 right, but not really..
```
Note: Since the FCnt field carries only the least-significant 16 bits of
35 the 32-bits frame counter, the server must infer the 16 most-significant
36 bits of the frame counter from the observation of the traffic.
```

giampiero7, 2021-07-10:
@flhofer  Not really. The full 32-bits counters are used for frame encryption and integrity check (see `4.3.3` and `4.4`). 
This is why `the server must infer the 16 most-significant bits`, to correctly decrypt and check the received frames.
Otherwise the possibility to use 32-bits frame counters would make no sense, don't you think?

flhofer, 2021-07-10:
It seems mostly related to frame uniqueness though. I think they didn't consider it as the toy-MKR micro might never reach millions of frames without any reset and probably (I think) doesn't have servers running in strict mode. 

Anyway, it seems that the firmware should handle it, so you can safely change the data types of getters and setters to uint32_t.

flhofer, 2021-07-10:
Oh, and sorry @giampiero7 if I sounded rude. I'm a bit under pressure here as I'm using these toys for a project and this modem firmware is really driving me nuts. We're discussing replacement so if you have some other suggestions, please come forward.

giampiero7, 2021-07-10:
@flhofer Not at all, on the contrary, I appreciate you replaying to my issue :)

You may want to check out my company's products:
https://www.sferalabs.cc/product/iono-mkr/
which are actually used in contexts where devices are possibly never reset and strict mode is very likely.
The core is a standard MKR, so adapting your current code would require minimal or no modification at all, depending on what you are doing.

Also, you can check our version of the library in this application:
https://github.com/sfera-labs/iono-mkr-lorawan/blob/master/IonoMkrLoRaWAN/MKRWAN.h
where we've modified it exactly as you suggested.

Hope it helps!

flhofer, 2021-07-10:
Awesome nice tools :) Finally something not Pi-based.
You might like to hear that I implemented a lot of patches and features for the firmware and the library. You can find them in the pull requests I submitted. Once I'm done with my pen- and performance testing, we might add final fixes and merge.

giampiero7, 2021-07-12:
Great, I'll check them out, thanks for your contribution. Good luck with your project!