arduino-libraries/MKRWAN #57: "FirstConfiguration" ABP not working

TechNyquist, 2019-07-26:
Hello everyone,

First things first: Arduino MKRWAN 1300, firmware just updated to 1.1.9, and MKRWAN library 1.0.10 .
For me ABP join using bare "First configuration" example fails, while OTAA 1.0 works nicely.
Is it just me or that happens to you too?

facchinm, 2019-07-30:
Hi @TechNyquist ,
this is quite strange... Which network are you trying to connect to?

TechNyquist, 2019-07-30:
Hello @facchinm ,
I'm using a local dev LoRaServer framework, with two gateways.

facchinm, 2019-07-30:
It doesn't indeed. We tested ABP with TTN and the Arduino Pro gateway (A2A network) and they both work correctly. Probably it's a LoRaServer issue? [This](https://github.com/brocaar/loraserver/issues) should be the right place to report it

TechNyquist, 2019-07-30:
Ok, thank you for your feedback (LoRa is pretty difficult to debug), I'll ping them.
I'm closing this issue and I'd reopen it whenever should the lights fall back here again.
Good job.

closed 2019-07-30 by TechNyquist

TechNyquist, 2019-08-02:
Ok, sorry everyone, it was my fault.
I was facing a simple frame counter validation issue. Just needed to set the right counter value with "setFCU" and all worked. Thanks and sorry.

For anyone facing this on the same environment, I redirect you to [this LoRaServer forum post](https://forum.loraserver.io/t/error-device-session-does-not-exist-or-invalid-fcnt-or-mic/379/30) that might give you all the required clues to understand and solve why ABP activation is not working.