arduino-libraries/MKRWAN #58: Set transmission frequency of MKRWAN

deanpretorius, 2019-07-28:
I have a setup consisting of a Dragino LoRa HAT on a Raspberry PI 3b+, and two Arudino MKR WAN 1300 devices. After installing and running the dual_chan_pkt_fwd software, the gateway was successfully registered and connected to TTN. Sending messages from the Arduinos are the problem. Running the example scripts for the MKR WAN devices, with the data rate set to 5(SF7) and the ADR set,  allowed me to send data to TTN through the Dragino HAT. This is a single channel Gateway. So the messages were all sent on SF7. However, the data passed through the hat is often corrupted. I believe that this is due to the frequency hopping of the MKRWAN. My devices are the only LoRa devices in the area, and there is no obstructions between the devices. Even sending “Hello” to TTN returned strange values in the payload. However, it would work in irregular intervals. [](url)
After looking deeper into the issue, I tried communicating on the simple LoRa network, without any LoRaWAN stuff, just using the arduino-LoRa library on SF7 and BW125. I noticed that the Arduinos could chat easily between each other. But they struggled to communicate with the Dragino device. This is the output of RPI receiving data from the arduino. The sent message is “hello” plus a counter. 
However, I am unsure how to fix this issue. All devices are communicating at 868.1MHz at SF7. Or it could be that the Arduino is changing its transmission frequency in a form of frequency hopping. Has anyone experienced anything similar to this and would be able to offer advice? Is there a way of setting the transmission frequency on the MKRWAN to prevent this frequency hopping?
I had a look at the issues in the repositories, and am beginning to understand the limitations of the products I have acquired. But most of the issues of using the MKRWAN with a single channel gw was closed last year. I hope that these issues have been solved since them.

facchinm, 2019-07-30:
Hi @deanpretorius ,
were you using https://github.com/sandeepmistry/arduino-LoRa for simple LoRa communication? In that case there should be no channel hopping unless you select it explicitely

deanpretorius, 2019-07-30:
Hi @facchinm
That is the library I used when I noticed the corrupted incoming messages. In that library I set the dataRate(5), SF7BW125. When I call getDataRate after that, it returns a value of -1. But I assume the dataRate does get set because the gateway receives messages, however, corrupt as described above. However, at present, there is no method available to specifically set the transmission frequency or to view the transmission frequency.

cliffi30, 2019-12-15:
Hello facchinm,
first of all thanks for all your support in the community around this topic and others. I am facing the same issue that deanpretorius is facing. I am trying to make MKRWAN1310 to transmit only on one single frequenc so I can fetch every transmisstion with my LG02. I didn't find a way to set this via the Modem class. I was trying to set it up using the info from your earlier recommended way [https://github.com/sandeepmistry/arduino-LoRa]) but that didn't work either. Which part on that repository are you actually refering to? --> Lora Simple Node?

facchinm, 2019-12-16:
Hi @cliffi30 ,
LG02 and other single frequency "gateways" are just simple radios that do not implement the full LoRaWAN protocol, so the best way to communicate with them is via plain LoRa (using Sandeep's library, Simple Node example is fine), grab the messages on the gateway side and route them to the internet (without relying on The Things Network or other providers that need full LoRaWAN compliance)