arduino-libraries/MKRWAN #59: Setting TX power always returns false

TechNyquist, 2019-07-30:
Hello eveyone,

I see that with set TX power method "power", trying every possible combination of mode and level it actually always returns false. Tried both after and even before activation, same.
I was following the ISSUE #46 and the "volatile" status of the library caught my attention and actually I got lost in the end.
Now what should I think about this?
- expected behaviour as library is going to be rewritten?
- don't worry about the "false" as it is actually working anyway?
- don't care about that method as it's no more maintained and will explode in few seconds?
- well, shouldn't return "false", so there's something wrong?
- whatever else?

Please, shed some light on my way, thanks.

nrcrast, 2021-01-26:
I am running into the same issue currently. I added a getter to grab the RFPOWER and it's always 0 no matter what I do. Any ideas anyone?

flhofer, 2021-07-08:
The problem has been solved with the latest [Firmware](https://github.com/arduino/mkrwan1300-fw/pull/35) and [library](https://github.com/arduino-libraries/MKRWAN/pull/93). Should be merged soon to master.