arduino-libraries/MKRWAN #60: Does void setBaud(unsigned long baud) work?

sslupsky, 2019-07-31:
I have attempted to use the setBaud() function provided in the library.  I wasn't able to change the baud rate and when I looked further into the mkrwan1300-fw library, I could not find the corresponding "+UART=" AT command.  So I suspect this command is being ignored by the murata module.

Can anyone else confirm if this is the case?

sslupsky, 2019-07-31:
Ultimately, what I am attempting to establish is if the baud rate between the SAMD21 and the Murata module can be increased to 115200 from the present 19200.  If I manually change the baud rate here https://github.com/arduino-libraries/MKRWAN/blob/2fa519d6bd8aceb359e3c3fcff91f0bd4b4324be/src/MKRWAN.h#L407 the module doesn't initialize.

facchinm, 2019-08-01:
Confirmed, AT+UART was only available in Murata FW and has not been ported (yet) https://github.com/arduino/mkrwan1300-fw/blob/master/Projects/Multi/Applications/LoRa/AT_Slave/src/command.c .
The modification to run at 115200 is trivial btw if you can compile a mkrwan1300-fw project
Just need to change this line https://github.com/arduino/mkrwan1300-fw/blob/188a5cd9b595d342b6c00d3dc9e9b9f7a102ef6b/Projects/Multi/Applications/LoRa/AT_Slave/src/vcom.c#L117

sslupsky, 2019-08-01:
Thank you for the link to the point in the code where the baud rate is set ... I went looking for that yesterday and could not locate it.  This will help to reduce power consumption since the module is polled when it wakes up because downlink messages are dropped if received during sleep.  Increasing the baud rate of the serial port accordingly reduces the time it takes to poll the module.

The UART can wake the SAMD21 from sleep but to do so requires that the clock source for the SERCOM peripheral be reconfigured to enable it during standby.  Presently, GCLK0 is used as the source for the SERCOM's which is the DPLL48 48MHz clock.  So, I don't think I want that one running during sleep. To achieve a higher baud rate, the module would need to be clocked by the OSC8M source.  If I do so I expect that the power consumption will increase about 65uA according to Table 37-12 in the datasheet.  

If I poll once every couple minutes the average current of the poll is about 1 uA.  This will be ok for my class A device.  A class C device would be problematic though.

closed 2019-08-01 by sslupsky