arduino-libraries/MKRWAN #71: ARD-078 1.2.0 and ABP

Dokhra, 2019-12-07:
Hello,

iam using the MKRWAN1300 with firmware 1.2.0 with a Dragino LG02 Gateway and the FirstConfiguration.ino (868100000Mhz).
With OTAA i can see join requests but with ABP it comes nothing to TTN.
Are there any ABP Issues with the 1.2.0 Firmware?

Best Regards,
Martin

facchinm, 2019-12-09:
Hi @Dokhra ,
there's no know issue on ABP for firmwares > 1.1.2 .
I just tested 1.2.0 with ABP on TTN (using a Raspberry Pi with IMST iC880A as gateway) and it works as expected...
Can't you see any activity on the TTN backend?

Dokhra, 2019-12-09:
Hi @facchinm 

I can't see any message with ABP on TTN. Only with OTAA i can see join requests, but no uplink.
Ttoday i read the Dragino LG02 Gateway manual and its recommend to use it NOT for LoraWAN due limitations (whats your opinion to that):

1/ It works only on one frequency at a time
2/ It works only for one DR at a time
3/ LoRaWAN compatible issue
4/ Downlink & OTAA issue

*Limitation of LoRaWAN support please refer section 3.6.2 of:
https://www.dragino.com/downloads/index.php?dir=LoRa_IoT_Kit/v2-Kit/&file=Single%20Channel%20LoRa%20IoT%20Kit%20v2%20User%20Manual_v1.0.1.pdf

So i decided to buy a TTIG-868 Indoor Gateway and make a external Antenna on it.

Best Regards,
Martin

facchinm, 2019-12-09:
Yes, single channel gateways are not real LoRaWAN gateways, so the limitations are heavy. A multichannel indoor gateway (the cheapest one is [this](https://www.thethingsnetwork.org/docs/gateways/thethingsindoor/) ) works mush better :slightly_smiling_face:

closed 2019-12-09 by facchinm

Dokhra, 2019-12-12:
Hi,

just want to give you a short feedback. Today I started to use the TTIG-868 Gateway and now ABP works without any issues.

Best Regards,
Martin