arduino-libraries/MKRWAN #74: using setFCU

radioisotopes, 2020-01-26:
I'm saving the state of a node between deep sleep

when I reinstate the value of framecounter using setFCU the framecounter doesn't advance normally after data uploads. framecounter gets "stuck" at the value I reinstated and then causes sequence errors at the network server.

UNLESS, i ask for the ack message after every upload 👍 
 err = modem.endPacket(true);

that's not the behaviour I expect, I shouldnt have to ask for an ack message to restart the framecounter after using setFCU.

any experiences of this?

xoseperez, 2020-07-23:
Not my experience. I'm using a TPL5111 to turn off completely the mkrwan1300. Before killing it I'm saving its state in an ATECC508A slot. After boot I recover the values from the slot and use the setFCU to set the counter and it works, no ACK required.

aalbinati, 2020-09-27:
@radioisotopes I'll be testing out this issue during the week. If I have the same problem I'll let you know