arduino-libraries/MKRWAN #75: joinOTAA not working on 1310 and US915

k8rapps, 2020-03-17:
I'm trying connect my MKRWAN 1310 to The Things Network.  I have a device configured for OTAA setup in the TTN and when I run FirstConfiguration and enter the APP EUI and APP Key, I actually see the device connect in the TTN console immediately, but after about 30 seconds, joinOTAA returns 0.

facchinm, 2020-03-18:
Hi @k8rapps ,
did you try `US915_HYBRID` band? The MKRWAN firmware is not super up-to-date and the regional specifications have slightly changed in the meantime.
The behaviour you are experiencing could be due to the gateway responding to OTAA on a sub-band you are not listening to.

k8rapps, 2020-03-18:
Hi @facchinm 

Thanks for the quick response.  I tried 915_HYBRID but I am seeing the same behavior.  Since the MKRWAN want firmware isn't up to date, Is there a different library I should be using?

Or do you mean the firmware on the modem?

facchinm, 2020-03-18:
Unfortunately it's about the modem firmware :unamused: 
I collected some ideas here https://github.com/arduino-libraries/MKRWAN/issues/64 on how to proceed, but no time has been spent yet in the task

k8rapps, 2020-03-18:
Gotcha, guess I'll use a different board in the meantime.

k8rapps, 2020-03-18:
If it helps, my gateway is an RAK2245 Version B

egionet, 2020-03-19:
Wondering if I am having the same problem.  I have an Dragino LPS8 Gateway and MKR WAN 1310 both set to US915.  I am just using the 'FistConfiguration' sample to simply test connectivity which is failing.  Dragino support came back with the following; "Please make sure the MKR use the same subband as LPS8 . The US915 band has 8 sub band and the LPS8 use subband 2 as default.".

javafrog, 2020-04-08:
I'm unfortunately experiencing the same problem. The joinOTAA returns false, but even after that, I can see the device trying to join in the TTN gateway traffic inspector =(

marekpetrik, 2020-08-25:
I have had the same problem. I am using MKR WAN 1310 and have the TTN indoor gateway (in the same room). joinOTAA works for me intermittently. The problem seems to be that even though the join is accepted (as I can see from the TTN gateway console), the modem does not appear to receive the acknowledgment. I wonder if it is related to this issue (for a different hardware):

https://github.com/matthijskooijman/arduino-lmic#problems-with-downlink-and-otaa

In any case, adding a delay(5000) between modem.begin(US915) and modem.joinOTAA(..) makes joining much more reliable. I am not sure why.

digamesystems, 2021-01-01:
> I have had the same problem. I am using MKR WAN 1310 and have the TTN indoor gateway (in the same room). joinOTAA works for me intermittently. The problem seems to be that even though the join is [...]

I'm having the same issues as you, Marek. -- Intermittent connection success to a TTN indoor gateway from across my house using otaa. I do see messages from the device in the TTN console as it attempts to connect but often modem.joinOTAA fails.

Any updates to this bug? I'm using MKRWAN.h version 1.0.13. My MKRWAN 1310 returns: 

"Your module version is: ARD-078 1.2.0"

Anyone have better luck with ABP activation?

facchinm, 2021-01-05:
Hi @digamesystems ,
the gateway is very likely "rejecting" some of the bands being configured by `US915`.
You can try setting `US915_HYBRID` which restricts the channels.
We are actively working on a solution to land on fw 1.3.0 (with all the new regional specifications fixed)

FarmerJD28, 2021-01-06:
I also have the MKRWAN 1310 using RAK7258 and TTN. My problem is that there is never a join that I can see on TTN in the GW traffic. I have tried FirstConfiguration and LoraSendAndRecieve (modified for my HW) and both never get past this line in the code:
`  int connected = modem.joinOTAA(appEui, appKey);`
It just hangs somewhere in there and never returns to my sketch.
I have also tried US915_HYBRID.
It would be a good idea if that join code would provide feedback instead of just hanging forever.
Any date for fw 1.3.0? Any idea if this problem would be fixed?
Thanks.

silverio, 2021-02-21:
I am having the same issue, any updates on this?

hpssjellis, 2021-03-01:
I am having the same issue with the US915 but with the new Lora Vision Shield. Any updates would be appreciated. I will try the delay(5000); after the begin(US915)

silverio, 2021-03-01:
I am using RAK7258 and a MKRWAN 1310. After trying everything this is what I did to make it work.

1. I updated the MKRWAN library to version 1.0.15
2. Updated the MKRWAN FW using the MKRWANFWUpdate_standalone sketch
3. Use de following in your sketch: `modem.begin(US915_HYBRID);`
 `modem.joinOTAA(appEui, appKey);`
 `modem.setPort(10);`
  `modem.dataRate(3);`
  `modem.setADR(true);`
4. Update the RAK7258 to the latest FW
5. Change frequency plan to sub-band  channel 0~channel 7, channel 64

After this, my arduino connects every time without issues.

hpssjellis, 2021-03-01:
Thanks @silverio my issue is that I never get past modem.joinOTAA(appEui, appKey); 

I have heard about the 2 settings. Thanks for confirming that.
modem.setPort(10);
modem.dataRate(3);

Can you get any connection using US915 ?

I don't have the RAK so can't edit my Gateway as well.

silverio, 2021-03-01:
@hpssjellis the issue I had with US915, was that sometimes it connected and sometimes it didn't. It wasn't reliable.

hpssjellis, 2021-03-01:
On the concept of things working badly using US915, anyone know why this [line of code is set](https://github.com/arduino-libraries/MKRWAN/blob/85b458486f277260c821357720abeda357025938/src/MKRWAN.h#L464-L466) to only work for EU868?

```
    if (band == EU868 && isArduinoFW()) {
        return dutyCycle(true);
    }

```

My problem testing US915 is I am not sure if my Gateway is any good. If I could get point-to-point LoRA working on the LoRa Vision Shield then I can find out what works without relying on a Gateway. Just a thought.

FarmerJD28, 2021-03-02:
> I am using RAK7258 and a MKRWAN 1310. After trying everything this is what I did to make it work.
> 
> 1. I updated the MKRWAN library to version 1.0.15
> 2. Updated the MKRWAN FW using the MKRWANFWUpdate_standalone sketch
> 3. Use de following in your sketch: `modem.begin(US915_HYBRID);`
>    `modem.joinOTAA(appEui, appKey);`
>    `modem.setPort(10);`
>    `modem.dataRate(3);`
>    `modem.setADR(true);`
> 4. Update the RAK7258 to the latest FW
> 5. RAK7258 - Change frequency plan to sub-band  channel 0~channel 7, channel 64
> 
> After this, my arduino connects every time without issues.

Silverio, Thanks for your input.
I also have the RAK7258 and registered on TTN. I have set everything as you described and still my LoraSendAndRecieve sketch hangs on the join statement except on very rare occasions when it joins and I can send/rec data (once it joined after sitting there for 7 minutes). Is your module version ARD-078 1.2.1?
Is there anything else you can think of in your sketch or gateway that might be key?
I have tried various places from the same room to adjacent room from the gateway.
I have been working with arduino and they have not been particularly helpful.

silverio, 2021-03-02:
Hi @FarmerJD28, yes I have the same module version. First, I did some tests with the node and the gateway in the same room, until I got a successful connection every time (using the config in the previous post). When I knew this connection was reliable, I then moved the node to another room, and got a reliable connections at a greater distance. At last I went to my farm and did some long range tests. I raised the gateway antenna 7 meters high, and then moved the node to multiple locations around the farm and got the RSSI and SNR. All readings came up in to the gateway without issues. The farthest test I did was at 2km. I am using the 8dbi antenna from RAK on the gateway, and on the node the included antenna.

When the gateway receives the data, it sends it to my server using the uplink data URL on the integrations tab. 
Everything I needed the gateway did it, so I decided not to use TTN. My gateway is in Network Server mode. Here is the default config.

[image]

Here an example of the sketch I was using for my tests:

```#include <MKRWAN.h>
#include "ArduinoLowPower.h"
#include <CayenneLPP.h>

LoRaModem modem;

#include "arduino_secrets.h"
// Please enter your sensitive data in the Secret tab or arduino_secrets.h
String appEui = SECRET_APP_EUI;
String appKey = SECRET_APP_KEY;

int long_sleep = 60000 * 2;
int short_sleep = 60000 * 1;

CayenneLPP lpp(51);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);
  // change this to your regional band (eg. US915, AS923,US915_HYBRID ...)
  if (!modem.begin(US915_HYBRID)) {
    Serial.println("Failed to start module");
    while (1) {}
  };

  Serial.print("Your module version is: ");
  Serial.println(modem.version());
  Serial.print("Your device EUI is: ");
  Serial.println(modem.deviceEUI());

  int connected = modem.joinOTAA(appEui, appKey);

  Serial.println(connected);

  if (!connected) {
    Serial.println("Something went wrong; are you indoor? Move near a window and retry");
    while (1) {}
  }

  modem.setPort(10);
  modem.dataRate(3);
  modem.setADR(true);

}

void loop() {

  Serial.println("delay 5 seconds");
  delay(5000);

  int err;
  int sleepTime = long_sleep;

  // Prepare Cayenne LPP
  lpp.reset();
  lpp.addAnalogInput(1, -18);
  lpp.addAnalogInput(2, -46);
  lpp.addAnalogInput(2, 80);

  modem.beginPacket();
  modem.write(lpp.getBuffer(), lpp.getSize());

  err = modem.endPacket(true);
  if (err > 0) {
    Serial.println("Message sent correctly!");
  } else {
    Serial.println(err);
    Serial.println("Error sending message :(");
    Serial.println("(you may send a limited amount of messages per minute, depending on the signal strength");
    Serial.println("it may vary from 1 message every couple of seconds to 1 message every minute)");
  }
  delay(5000);
  USBDevice.detach();
  LowPower.deepSleep(sleepTime);
  USBDevice.attach();

}

hpssjellis, 2021-03-02:
Thanks @silverio and @FarmerJD28 really interesting that it is working for you without TTN but using the network server mode. I hope someone else with the Lora Vision Shield joins the conversation. I am putting my code on my Portenta site if anyone wants to comment or send a PR. Might code hopefully will let me drive closer to a solid gateway and see if my devices connect. The LED's let me know what is happening without a serial connection.

[research/LoRaWan](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/research/LoRaWan)

FarmerJD28, 2021-03-03:
> Hi @FarmerJD28, yes I have the same module version. First, I did some tests with the node and the gateway in the same room, until I got a successful connection every time (using the config in the [...]

Thanks for all the input @silverio. I tried all that you posted with no luck other than a few joins with countless attempts like before. If I get this working I may also use the gateway as a Network Server instead of TTN. Thanks for pointing that out. 
At this point Arduino says they will send me a new MKR1310. So we will see if I had a marginal board but I am not holding my breath.
Good luck @hpssjellis with your Lora Vision Shield.

hpssjellis, 2021-03-08:
By the way @FarmerJD28 and @silverio I had no trouble connecting to the https://explorer.helium.com/coverage [ThePeoplesNetwork](https://www.helium.com/) I have not yet sent a packet but at least I got connected. Too bad the nearest gateway is 40 km away.

FarmerJD28, 2021-03-30:
I resolved my join problems!
I live in snow country so my house has a metal wainscot around the base of the house. My gateway was in the house but near the outside wall. My MKR1310 was also inside the house and the metal wainscot WAS NOT between the two. I suspect that the metal was causing some reflections or other distortions to the signal. When I move the gateway and the MKR1310 up to the 2nd floor I have 100% connection rate and successful up/down links.

rpruizc, 2021-04-16:
> Thanks @silverio and @FarmerJD28 really interesting that it is working for you without TTN but using the network server mode. I hope someone else with the Lora Vision Shield joins the conversation. [...]

Another 1310/Vision frustrated user. Neither device is able to joinOTAA my TTN gateway. Just tried with the newest firmware version for MKRWAN v1 and v2 (1.1.0 & 1.3.2). Same result: "Something went wrong; are you indoor? Move near a window and retry"
Sadly I don't have another gateway to test this one out, and I'm kind of anxious about investing another couple hundred dollars on another gateway just to see the 1310 nor the vision being able to join. This is super frustrating.

hpssjellis, 2021-04-17:
@rpruizc I just did some testing with my Portenta Vision Shield on the  Helium network and still nothing with my MKRWAN library. I know with TTN an Arduino employee who got it working but only with the gateway on the new V3 servers. Here is the Arduino [tutorial](https://www.arduino.cc/pro/tutorials/portenta-h7/vs-ard-ttn)  Of course I am not able to migrate my Gateway from V2 to V3 so I can't test out what the tutorial does. If I could set it up the new servers even identify the Portenta as a known product which is encouraging. 

I do have a solution for the Portenta but it is a bit hacky using the GrumpyOldPizza library, presently trying to communicate with the murata module because this solution is useless unless I can communicate with it. Not sure if it would work for the MKRWAN 1300. I will get back if I have any success.

hpssjellis, 2021-04-20:
Wow @rpruizc, so many other products. I fully understand. Good luck.

dkilcy, 2021-08-14:
Same problems here with the 1310 (ARD-078 1.2.0) and Dragino LG308 with TTNv3.   I wasted all day trying to get this working.  With the issue opened over a year to fix basic functionality, it doesn't give me hope.

rqg0717, 2022-01-26:
I am having the same issue with a Dragino LPS8 Gateway and MKR WAN 1300. 
I am using the following in the example sketch: 
```
modem.begin(US915_HYBRID);
modem.joinOTAA(appEui, appKey);
modem.setPort(10);
modem.dataRate(3);
modem.setADR(true);
```
Here is the serial port print:
```
Your module version is: ARD-078 1.2.3
Please make sure that the latest modem firmware is installed.
To update the firmware upload the 'MKRWANFWUpdate_standalone.ino' sketch.
Are you connecting via OTAA (1) or ABP (2)?
Enter your APP EUI
Enter your APP KEY
Something went wrong; are you indoor? Move near a window and retry
```
MKRWAN lib version is 1.1.0

Any thought? Thank you in advance.

hpssjellis, 2022-01-27:
@rqg0717 Hey James. I have a few programs you could try, they are made for the PortentaH7 US915 on the Helium network (The correct channel needed to be enabled: 1 for Helium and I think 0 for TTN) but might work on the mrk1310.

https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot3-portenta-vision-shields/dot33-lorawan-specific

The library can be installed searching for community and installing portenta-pro-community-solutions

Can you tell me if any of them work since I don't have your board. 

All of them need you to run the MRKWAN standalone program with the serial monitor to install it.

rqg0717, 2022-01-27:
@hpssjellis  Thank you so much. I will try to enable the correct channel for TTN. I was wondering where I could find the definition of the channels. Thank you.

johnstaveley, 2024-09-19:
I am having this problem.
Using:
MKRWAN_v2 library
_lora_band region = EU868;
Module version is: 1.3.0.0
Mkr Wan 1310 hardware 
I can see the joins in TTN:
Accept join request
Successfully processed join request
Forward join accept message

but the MKR wan reports "Something went wrong; are you indoor? Move near a window and retry"