arduino-libraries/MKRWAN #79: MKR WAN 1310 sleep current

zuyan9, 2020-05-19:
Hello guys,

On the MKR WAN 1310 [product page](https://store.arduino.cc/usa/mkr-wan-1310), it says
> When properly configured, the power consumption is now as low as 104uA!

However I am not able to achieve that, I got around 450uA by cutting the jumper and using LowPower.deepSleep(), similar to the results in [this discussion](https://forum.arduino.cc/index.php?topic=644957.0).

Is there any way to achieve lower sleep current?

Thanks

zuyan9, 2020-05-21:
By changing the pin mode from `INPUT` to `INPUT_PULLUP` in [wiring.c](https://github.com/arduino/ArduinoCore-samd/blob/46f6cad46b55aab1e6081bc1bb60e4fd66dad4c0/cores/arduino/wiring.c#L89), the sleep power was reduced to ~160uA. Anything else we can do?

facchinm, 2020-05-22:
Hi @zuyan9 ,
your modification in wiring.c should not affect the MKRWAN since board defines VERY_LOW_POWER (and that code doesn't get compiled / called). Is it possible that you are doing other things in your sketch that influence the consumption?

zuyan9, 2020-05-22:
Hello @facchinm 

You are right, my bad, for MKR WAN 1310 `VERY_LOW_POWER` is defined, that code won't get compiled, I picked MKR WAN 1300 by mistake.

Using [TimedWakeup](https://github.com/arduino-libraries/ArduinoLowPower/blob/master/examples/TimedWakeup/TimedWakeup.ino) example in ArduinoLowPower, I am able to achieve 15uA sleep current.

I played with pinMode a bit and imported this sketch to Atmel Studio 7 too. The measured sleep current results are interesting.

pinMode | Arduino IDE |  Atmel Studio | 
---|---|---|
Nothing | 15uA | 15uA |   
INPUT | 550uA | 550uA |   
INPUT_PULLUP | 155uA  | 15uA |   
INPUT_PULLDOWN | 155uA  | 155uA |   
OUTPUT | 155uA  | 1390uA |   

I don't know why there are huge differences in some cases, the sketch code and ArduinoCore-samd are exactly the same, maybe difference in the bootloader?

sslupsky, 2020-07-12:
@zuyan9 those look like mkr wan 1300 numbers unless you made a hardware modification to the 1310?

zuyan9, 2020-07-12:
> @zuyan9 those look like mkr wan 1300 numbers unless you made a hardware modification to the 1310?

This is for MKR WAN 1310, I did cut the solder jumper under the battery connector and supply external 3.3V.

amasidlover, 2021-02-10:
I'm using an MKRWAN 1310 with jumper cut, with the following sketch:
`#include "ArduinoLowPower.h"
#include <LoRa.h>
#include <MKRWAN.h>

LoRaModem modem;

void setup() {

   for (int i=0; i < 15; i++) {
   pinMode(i, INPUT_PULLUP);
 }
  pinMode(LED_BUILTIN, OUTPUT);

    digitalWrite(LED_BUILTIN, HIGH);
  delay(3000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);

 
}

void loop() {
    USBDevice.detach();

  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);

    LowPower.deepSleep(8000);

  LoRa.begin(868E6);
  LoRa.beginPacket();
 LoRa.print("D:");
  LoRa.endPacket();
  LoRa.end();
  
  digitalWrite(LORA_IRQ_DUMB, LOW);
  digitalWrite(LORA_RESET, LOW);
 digitalWrite(LORA_BOOT0, LOW);
 
}`

I get a sleep current of 16uA (!) on the first sleep, however, after the LoRa send has occurred the sleep current sticks at 300uA. I've tried a variety of things to get the LoRa modem / Murata unit to shut down ( modem.dumb(), modem.sleep() ) but nothing seems to cause it to come back to the state it was in when the board is first powered on. 

The data sheets say that deep sleep should only be a small number of uA.

I've also found the 'Grasshopper' boards (too late for this project as I now have 6 MKRWAN 1310s...) which directly control the stm32 within the MuRata module and they explicitly do an stm32.stop() after sending and then see 2-3uA sleep currents from it.

At this point I'd be happy with the 100uA sleep talked about up thread. 

I think what is needed is for the MKRWAN library to either be able to send an AT command that triggers deep sleep (modem.sleep() appears to hang my board and leave it using 20mA) or attach to the STM32 and cause it to 'stop' - unfortunately I can't find a complete set of AT commands for the unit to get further (I suspect Murata control access to those who have the dev kit for the module) and I'm really struggling to see how direct commands to the STM32 work - although I can see that the firmware updater does this.

zuyan9, 2021-02-10:
@amasidlover yes 16uA of sleep current should be achievable. 

The firmware for the module can be found in [mkrwan1300-fw](https://github.com/arduino/mkrwan1300-fw), it's based on (a rather old version of) ST's [I-CUBE-LRWAN](https://www.st.com/en/embedded-software/i-cube-lrwan.html) with some modifications. It is possible to figure out the AT commands by reading the code (or refering to I-CUBE-LRWAN manual, but some commands have changed).

Murata has their own module firmware too and they have a different set of AT commands. It is not publicly available and it's not used here for Arduino.

I don't think the sleep current variation is due to the module, I suspect it has something to do with ArduinoCore-samd but have yet to identify that.

masa8, 2022-01-30:
I get a sleep current of 60uA even after using LoRa by MKRWAN 1310 with jumper cut, with the following sketch:

```
#include "ArduinoLowPower.h"
#include <MKRWAN.h>

LoRaModem modem;

void setup() {
  
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  pinMode(1, INPUT_PULLUP);
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  pinMode(7, INPUT_PULLUP);
  pinMode(8, INPUT_PULLUP);
  pinMode(9, INPUT_PULLUP);
  pinMode(10, INPUT_PULLUP);
  pinMode(11, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
  
  //modem.sleep();
  USBDevice.detach();
  
}

void loRaThing() {
  modem.begin(EU868);
  modem.beginPacket();
  modem.print(":D");
  modem.endPacket(true);
  modem.sleep();
  
}
void loop() {
  loRaThing();
  LowPower.deepSleep(10000);
}
```

etienneesp, 2022-05-13:
Try to add this lines : For me, it's works and I save 300 µA !

pinMode(LORA_IRQ_DUMB,INPUT);
  pinMode(LORA_RESET,INPUT);
  pinMode(LORA_BOOT0,INPUT);

My send function :

void LoRa_sendMessage(String message) {
  LoRa.begin(868E6);
  LoRa.beginPacket();
  LoRa.print(message);
  LoRa.endPacket();
  LoRa.sleep();
  digitalWrite(LORA_IRQ_DUMB, LOW);
  digitalWrite(LORA_RESET, LOW);
  digitalWrite(LORA_BOOT0, LOW);
  pinMode(LORA_IRQ_DUMB,INPUT);
  pinMode(LORA_RESET,INPUT);
  pinMode(LORA_BOOT0,INPUT);
}

Dicko87, 2022-12-11:
Hi guys, hi @zuyan9 did you find a solution to this. I have a WAM1310, powered by a 3.7V LiPo battery and the sleep current is 4mA. Can you please tell
Me how to achieve the very low current consumption during sleep? I have not cut any solder link. I would then need a 3.3V regulator from the battery if I did that?
Any help would be great, thanks.

sslupsky, 2022-12-11:
@Dicko87 Using the on-board PMIC you should be able to achieve a sleep current a bit more than 100uA.  To achieve a better result requires you to disable the PMIC and supply power to the board using an alternate method.

Regarding your 4mA situation, you need to pay very close attention to two things:
1.  What MCU peripherals do you have enabled
2. Pin states

For instance, if you leave USB enabled you will not be able to achieve the low power numbers mentioned in this thread.

Similarly, the pin states can affect current consumption and causing unnecessary current to source or sink from an IO pin.

Dickoabc123, 2023-01-06:
@sslupsky thank you very much for your reply. Can you please tell me how I would use the on-board PMIC or point me to a resource that explains what to do.

Thank you very much.

LasaleFamine, 2023-09-20:
Would be great to have an example with the low power logic inside. Is there something out there?