arduino-libraries/MKRWAN #82: MKRWAN1300 FW stuck

gavriil07, 2021-01-16:
I'm trying to send sensor data using LoRA and MKRWAN1300. The LoRa Sender example fails to begin.
I read I need to do a FW first, but it gets stuck after I enter a key:
[image]

Any ideas?

zougloub, 2021-01-17:
I got that board for a friend and started playing with it before he does, noticed the same issue as you, so I ended up using the FWUpdaterBridge sketch and a modified stm32flash ( https://sourceforge.net/p/stm32flash/tickets/118/ ).

rogeralthaus, 2021-01-25:
I had the same problem...

After remove  of port.flush() line  the update was successful...
[image]

nrcrast, 2021-01-26:
I've also had success lowering the baud rate.

facchinm, 2021-01-28:
Hi everyone, the issue was indeed in `flush()` call. We released a fix by commenting that function (https://github.com/arduino-libraries/MKRWAN/commit/049391822099dae9db67a9ad11f4a2005e745547) but it will be solved at core leven in the next few days too.

closed 2021-01-28 by facchinm