arduino-libraries/MKRWAN #85: joinABP not working since library update from 1.0.14 to 1.0.15

wapel, 2021-02-16:
Using this command in my sketch on MKR1310:
```
  connected = modem.joinABP(devAddr, nwkSKey, appSKey);
  
  if (!connected) {
    if(debug)Serial.println("Lora ABP Connection failed !");
    while (1) {}
  } else {
    if(debug)Serial.println("Connected successful !");
  }
```

The "joinABP" is not working anymore since library update from 1.0.14 to 1.0.15
Perhaps the additional "sendAT()" is a problem ?
Thanks for help.

flhofer, 2021-04-26:
Same issue here, I observed the opposite problem. Updating firmware from 1.2.0 to 1.2.3 with the same 1.10 library and the thing stopped working. Must be a common change that was implemented in fw and library. There was something done to the default AT responses, I think.
[issue](https://github.com/arduino/mkrwan1300-fw/issues/33)

flhofer, 2021-04-28:
Fixed it on my side. The issue is that the new firmware responds according to standard with `+OK` on every command. So after the join event feedback, there is another `+OK` message in the queue. The status read comming next will then just read a `+OK` and hang forever for the `=` sign to read the join status. 

In line 950 of Mkrwan.h after the wait for `"+EVENT=1,1"` add
```
    if (waitResponse() != 1) {
      return false;
    }
```
This wait defaults to 1 second wait for `+OK` or `+ERR`

flhofer, 2021-04-30:
better solution #93. The `+OK` is due to an implemented hack but it doesn't seem to always show up. A now timed `skipUntil` should resolve the issue

flhofer, 2022-01-31:
See the not yet released [v1.2.4 of my pull request](https://github.com/arduino-libraries/MKRWAN/pull/93). It includes the necessary modem firmware update in the MKRWAN standalone sketch. (link to repo next to the title flhofer...)

Tocio88, 2024-07-23:
Hi, i'm using the library and when i do joinABP seems waiting forever for something. I'm using the firstConfiguration sketch.
When using the joinOTAA it works not problem, getting the data from the TTN console and parsing them to joinABP, i don't get any error and nothing happens, also in the TTN console no sign of data tx. I'm having my own gateway nextdoor, so no problem with the network i guess...