arduino-libraries/MKRWAN #88: Multiple compilation errors Redeclaration of 'CLASS_A', CLASS_B, CLASS_C

slowtoaccept, 2021-03-05:
Compiling this script from [https://www.arduino.cc/en/Tutorial/MKRWANFirstConfiguration]. Using US915. for MKR WAN 1300.
Get multiple errors for all classes, e.g., 
In file included from E:\Documents\Arduino\First_Configuration_MKRWAN\First_Configuration_MKRWAN.ino:4:0:

E:\Documents\Arduino\libraries\MKRWAN_v2\src/MKRWAN_v2.h:255:15: error: redeclaration of 'CLASS_A'

     CLASS_A = 'A',
For either MKRWAN_v2.h or MKRWAN.h . Using Arduino: 1.8.13 (Windows 10), Board: "Arduino MKR WAN 1300"

per1234, 2021-03-06:
Hi @slowtoaccept. Thanks for your report.

Please do this:
When you encounter an error you'll see a button on the right side of the orange bar in the Arduino IDE "Copy error messages". Click that button. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error in a message here using [code fencing](https://guides.github.com/features/mastering-markdown/).

slowtoaccept, 2021-03-06:
Thx for the fast response. Please see this: Arduino MKR WAN 1300 Compilation Error.docx

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Arduino MKR WAN 1300 Compilation Error.docx

 |

 |

 |

per1234, 2021-03-06:
@slowtoaccept GitHub does not accept email attachments. Please open the issue in your web browser:
https://github.com/arduino-libraries/MKRWAN/issues/88
Then post the output here directly on GitHub. Please don't use .docx format. If you want to do a file attachment, then it should be a .txt

slowtoaccept, 2021-03-06:
Here's complete error log. 

https://drive.google.com/file/d/1QOXHF0srtEDOKAuz82PEMcZmL65V-hjo/view?usp=sharing

slowtoaccept, 2021-03-06:
Here's complete error log in a .txt file.
[]https://drive.google.com/file/d/1t7RXj3rsHYa5a5kD_QrgyWNFAah1OWBC/view?usp=sharing

(https://drive.google.com/file/d/1t7RXj3rsHYa5a5kD_QrgyWNFAah1OWBC/view?usp=sharing)

per1234, 2021-03-06:
The relevant parts of slowtoaccept's output:
```
Arduino: 1.8.13 (Windows 10), Board: "Arduino MKR WAN 1300"
```
...
```
Detecting libraries used...

"C:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_SAMD_MKRWAN1300 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8053 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WAN 1300\"" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\cores\\arduino/api/deprecated" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\cores\\arduino" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\variants\\mkrwan1300" "C:\\Users\\slowt\\AppData\\Local\\Temp\\arduino_build_805479\\sketch\\First_Configuration_MKRWAN.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for config.h: [LoRa_Node@1.0.1]

ResolveLibrary(config.h)

  -> candidates: [LoRa_Node@1.0.1]

"C:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_SAMD_MKRWAN1300 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8053 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WAN 1300\"" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\cores\\arduino/api/deprecated" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\cores\\arduino" "-IC:\\Users\\slowt\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.11\\variants\\mkrwan1300" "-IE:\\Documents\\Arduino\\libraries\\LoRa_Node\\src" "C:\\Users\\slowt\\AppData\\Local\\Temp\\arduino_build_805479\\sketch\\First_Configuration_MKRWAN.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for MKRWAN_v2.h: [MKRWAN_v2@1.3.1]

ResolveLibrary(MKRWAN_v2.h)

  -> candidates: [MKRWAN_v2@1.3.1]
```
...
```
In file included from E:\Documents\Arduino\First_Configuration_MKRWAN\First_Configuration_MKRWAN.ino:4:0:

E:\Documents\Arduino\libraries\MKRWAN_v2\src/MKRWAN_v2.h:255:15: error: redeclaration of 'CLASS_A'

     CLASS_A = 'A',

               ^~~

In file included from E:\Documents\Arduino\libraries\LoRa_Node\src/mac/region/Region.h:56:0,

                 from E:\Documents\Arduino\libraries\LoRa_Node\src/LoRaNode.h:32,

                 from E:\Documents\Arduino\First_Configuration_MKRWAN\First_Configuration_MKRWAN.ino:2:

E:\Documents\Arduino\libraries\LoRa_Node\src/mac/LoRaMac.h:106:5: note: previous declaration 'eDeviceClass CLASS_A'

     CLASS_A,

     ^~~~~~~

In file included from E:\Documents\Arduino\First_Configuration_MKRWAN\First_Configuration_MKRWAN.ino:4:0:

E:\Documents\Arduino\libraries\MKRWAN_v2\src/MKRWAN_v2.h:256:5: error: redeclaration of 'CLASS_B'

     CLASS_B,

     ^~~~~~~

In file included from E:\Documents\Arduino\libraries\LoRa_Node\src/mac/region/Region.h:56:0,

                 from E:\Documents\Arduino\libraries\LoRa_Node\src/LoRaNode.h:32,

                 from E:\Documents\Arduino\First_Configuration_MKRWAN\First_Configuration_MKRWAN.ino:2:

E:\Documents\Arduino\libraries\LoRa_Node\src/mac/LoRaMac.h:112:5: note: previous declaration 'eDeviceClass CLASS_B'

     CLASS_B,

     ^~~~~~~

In file included from E:\Documents\Arduino\First_Configuration_MKRWAN\First_Configuration_MKRWAN.ino:4:0:

E:\Documents\Arduino\libraries\MKRWAN_v2\src/MKRWAN_v2.h:257:5: error: redeclaration of 'CLASS_C'

[... 16 lines cut ...]
In file included from E:\Documents\Arduino\First_Configuration_MKRWAN\First_Configuration_MKRWAN.ino:4:0:

E:\Documents\Arduino\libraries\MKRWAN_v2\src/MKRWAN_v2.h: In member function 'int8_t LoRaModem::waitResponse(uint32_t, arduino::String&, ConstStr, ConstStr, ConstStr, ConstStr, ConstStr, ConstStr, ConstStr, ConstStr)':

E:\Documents\Arduino\libraries\MKRWAN_v2\src/MKRWAN_v2.h:932:15: warning: unused variable 'port' [-Wunused-variable]

           int port = stream.readStringUntil(':').toInt();

               ^~~~

Using library LoRa_Node at version 1.0.1 in folder: E:\Documents\Arduino\libraries\LoRa_Node 

[... cut ...] ents\Arduino\libraries\MKRWAN_v2 
```
So this is a name collision between the MKRWAN library and the "LoRa Node" library (https://github.com/arduino-libraries/LoraNodeShield)

Regarding this:
```
Alternatives for config.h: [LoRa_Node@1.0.1]

ResolveLibrary(config.h)

  -> candidates: [LoRa_Node@1.0.1]
```
I don't see anything about config.h in the sketch or the MKRWAN_v2 library. So I guess you have modified the sketch and will be able to solve this by carefully copying the exact code this time.

closed 2021-03-06 by per1234

slowtoaccept, 2021-03-06:
Hey Per1234You wrote: "I don't see anything about config.h in the sketch or the MKRWAN_v2 library. So I guess you have modified the sketch and will be able to solve this by carefully copying the exact code this time." Please advise as to which version of FirstConfiguration.ino I should use:1) [https://www.arduino.cc/en/Tutorial/MKRWANFirstConfiguration]or 2) https://github.com/arduino-libraries/MKRWAN/tree/master/examples/FirstConfiguration 
They do not match. Look at lines 56-59.

 
 

Closed #88.