arduino-libraries/MKRWAN #89: Setting DR doesn't work

fe64970103, 2021-03-10:
It seems setting the data rate via dataRate() doesn't work at all and this is a known issue. There is a workaround I confirmed myself that works, but the workaround requires use of ADR that makes the gateway to send downlink messages on each and every uplink.

The workaround consists of enabling ADR and the setting the datarate prior sending out lora frames:
`modem.setADR(true);`
    `modem.dataRate(3);`
   ` modem.beginPacket();`
   ` ...`

However this doesn't seem to be a MKRWAN library issue, but someting related to the module itself. Is there something that can be done via raw AT commands to get around this? Any suggestions or clues are welcome.

fe64970103, 2021-03-13:
As already thought this seems to be a module firmware issue:
https://github.com/arduino/mkrwan1300-fw/issues/6

It's pretty old to and I'm wondering how such a basic feature wasn't addressed by now. This really limits the usage of the mkr1310 board as there is no real workaround.

flhofer, 2021-07-06:
@fe64970103 Check out the linked issue. Found the problem. Should be fixed (Patched) with the next version 1.2.4

flhofer, 2022-01-31:
See the not yet released [v1.2.4 of my pull request](https://github.com/arduino-libraries/MKRWAN/pull/93). It includes the necessary modem firmware update in the MKRWAN standalone sketch. (link to repo next to the title flhofer...)