arduino-libraries/MKRWAN #92: Endianness  AppEui

hpssjellis, 2021-04-14:
Endianness for the  AppEui:  most providers such as [TTN](https://www.thethingsnetwork.org/map) and [Helium](https://explorer.helium.com/coverage) have a default for Endianness but also a second method either LSB (least significant byte first) or MSB (Most significant byte first) . What is the expected Endianness for most Arduinos? I am testing:  [MKRWAN](https://github.com/arduino-libraries/MKRWAN)  ,  [MKRWAN_v2](https://github.com/arduino-libraries/MKRWAN_v2) and [GrumpyOldPizza](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0) . (Not really, I am actually using @facchinm GrumpyOldPizza board version for the stm32l0 core [here](https://github.com/facchinm/ArduinoCore-stm32l0) ) 

I don't think Endianness applies to the AppKey, but if you are planning on changing the DevEui Endianness would also be an issue.  Anyone have any opinions or at least what worked for them.

flhofer, 2021-04-26:
All LoRa libraries I used so far use big endian, i.e. copy the keys and EUIs as is into a string and it will work.

hpssjellis, 2021-04-26:
@flhofer Thanks for that information. Are you in North America or Europe or ... ? I am looking for simple TTN code that works in North America.  "I know it should just be changing to US915", but when things are not working it is nice to have a working bit of sample code.

flhofer, 2021-04-28:
I'm in Europe. Technically you should be able to "Just change to US915", all other settings are transparent. You might want to make sure that the firmware on the modem has not been compiled with a fixed regional band, but typically this is not the case.
Things might not work if your network has particular requirements. I seem to remember that US has in-transmission hopping as a mode. Don't remember if it was an option or a fixed thing.
Try the Send-receive sketch and let me know :)

closed 2021-05-31 by hpssjellis