arduino-libraries/MKRWAN #95: Unable to join through OTAA on EU868 spectrum

stormon-force, 2021-06-14:
I apologize if this is a duplicate, I am unable to see any join requests appear on my packet forwarder or network server when using the 1310 set to the EU868 region. I am fairly confident my gateway and server is setup correctly, as I have another device on the EU868 spectrum performing correctly. Is there anything extra I need to account for? This device may of previously been provisioned on the US915 spectrum. I have experimented with changing the data rate, enabling extra channels, and including a delay of ~5 seconds between setting the region and joining. 

MKRWAN version: 1.1.0
Murata module version: 1.2.3

Please let me know if there is anything I can try.

Code sample below
```/*

  Lora Send And Receive

  This sketch demonstrates how to send and receive data with the MKR WAN 1300/1310 LoRa module.

  This example code is in the public domain.

*/
#define LORA_DEBUG Serial

#include <MKRWAN.h>

LoRaModem modem;

// Uncomment if using the Murata chip as a module
// LoRaModem modem(Serial1);

//#include "arduino_secrets.h"
// Please enter your sensitive data in the Secret tab or arduino_secrets.h

String appEui = ""; // omitted in code sample
String appKey = ""; // omitted in code sample

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);
  // change this to your regional band (eg. US915, AS923, ...)
  if (!modem.begin(EU868)) {
    Serial.println("Failed to start module");
    while (1) {}
  };
  
  delay(5000);
  //Serial.print("Your module version is: ");
  //Serial.println(modem.version());
  //Serial.print("Your device EUI is: ");
  //Serial.println(modem.deviceEUI());
  //delay(10000);

  //Serial.println("Joining");
  int connected = modem.joinOTAA(appEui, appKey);
  if (!connected) {
    Serial.print("Something went wrong; are you indoor? Move near a window and retry ");
    Serial.println(connected);
    while (1) {}
  }

  Serial.println("Connected");
}

void loop(){
  delay(10000);
}
```

flhofer, 2021-07-06:
Can you set the LoraDebug serial to check what the SW is doing?
```
#define LORA_DEBUG 		Serial
```

stormon-force, 2021-07-07:
Yes, here you are. 

```### AT:
### AT: +DEV?
### AT: +VER?
Please update fw using MKRWANFWUpdate_standalone.ino sketch
### AT: +BAND= 5
### AT: +DUTYCYCLE= 1
Your module version is: ### AT: +DEV?
### AT: +VER?
ARD-078 1.2.3
### AT: +MODE= 1
### AT: +APPEUI= // appeui omitted
### AT: +APPKEY= // appkey omitted
### AT: +JOIN
### AT:
Something went wrong; are you indoor? Move near a window and retry 0
```

flhofer, 2021-07-08:
Ok, so it seems that it performs the normal Join procedure. I guess you have a 10-second pause between the last AT: and the message. The sketch and the behavior of the micro seem correct. Things that come to mind now are:

1. The App-EUI and APP-Key are not correct for your device: I have multiple devices on the same LoRa Server application and the same Network. While the Application ID (now called Join ID) stays the same, the Application key changes with the device EUI. Be sure these values are correct or the Join server/Network server will not answer
2. The antenna driver (power electronics) is damaged or the antenna is damaged causing very bad signal quality. I had a case where the detachable connector was defective.
3. The firmware on the device is compiled to use US915 regional settings only (custom compile?). As far as I know, that shouldn't be the case anymore
4. Is this all the code in the sketch? I had a buffer overflow in my code that caused sometimes the join to succeed, sometimes to fail.

Want to try the latest version of the library in my [pull-request?](https://github.com/arduino-libraries/MKRWAN/pull/93)

stormon-force, 2021-07-09:
Hi flhofer, thanks for your help. I believe the issue is 2., I wasn't using any antenna at all 😞. Attaching the antenna that came with the board allows me to join properly and see transmissions from the device. I incorrectly assumed because the 1310 worked on US915 fine without an antenna, the behavior would be the same for EU868. Is there a physical radio reason this would be a problem? Does it have to do with the frequency difference? The device is sending join requests but seems to have some trouble receiving join accepts. 

For gateways I am using a Multitech Conduit and RAK2245. Could there be an issue with gateways that might be set for US915? Both can receive and send on EU868 (I have other devices to confirm), but the Conduit gives a warning that there may be issues.

flhofer, 2021-07-10:
Yes, the US and EU use different signal coding, spread factors, have different bandwidths, and most important for your setup, US allows for up to 30dBm transmit power, while the EU only allows up to 20dBm (in exceptional cases). In general, I think the antenna should be used with the MKRWAN1300 as the modem was surely designed to operate with an antenna.

In cases where there is no antenna visible, it might be "hidden" in on the board. TTN uno's for example have the antenna drawn in the board by using [carefully placed PCB lanes.](https://www.thethingsnetwork.org/docs/devices/uno/led_on.jpg
)

closed 2023-03-01 by stormon-force