arduino-libraries/MKRWAN #100: ARDUINO_FW_VERSION version mismatch and does this library really support US915?

MichaelBla, 2021-08-24:
In the examples, the sketch that upgrades the firmware upgrades it to 1.2.3.

Then all the examples check to see that you're running 1.2.1 and give you a warning message if you're not.

More importantly, I can't get this to connect in the US although I can see it try in my gateway.

I can't find anyone else having success with this in the US so I wonder if it has been tested here.

MichaelBla, 2021-08-31:
> I am having success with the Helium network in North America using the new
> MKRWAN feature on the Portenta where you can enable channels:
>
> For TTN I think you have to enable the "0" channel, can someone check:
> '''
> modem.disableChannel(0);
> modem.enableChannel(1); // only one enabled for Helium
> '''
>
> Demo program here
>
> Here is the code I am using:
>
>
> /*
>   Helium Send And Receive
>   This sketch demonstrates how to send and receive data with the MKR WAN 1300/1310 LoRa module.
>   This example code is in the public domain.
>   note: Helium must be setup for what it does with the CayenneLPP encoded data
> */
>
> #include <MKRWAN.h>
> #include <CayenneLPP.h>
>
>
> [... 213 lines cut ...]
>   delay(30);
>   Serial.println("isChannelEnabled(6): " + String(modem.isChannelEnabled(6)));
>   delay(30);
>   Serial.println("-------------------------------");
> }

That worked!

In that I saw my first payload, ever. I was using US915 and enabled channel
0 and disabled channel 1.

I'll have to go through your code and figure out what it was trying to send
and if my decoded payload of "\u0001g\u0003!" is meaningful.

Sorry, I'm obviously new to this and having never received a payload before
I don't have any experience handling it.

I didn't wire-up any LEDs. I just assigned them to outputs 4-6.

Thanks! I finally feel like I have a foot in the door.

Michael

hpssjellis, 2021-08-31:
@MichaelBla Glad I could help,