arduino-libraries/MKRWAN #103: Is my MKR WAN 1300 not sending proper data to TTN?

OcimaSagledam, 2021-09-02:
Hello,
I have been having a problem with trying to send data to the LoRa network using the MKR WAN 1300. The code is:
#include <MKRWAN.h>
#include "arduino_secrets.h"
LoRaModem modem;

bool pump_running = false;
int water_level = 0;
const int empty_level = 0;
const int full_level = 5;

void setup() {
Serial.begin(115200);
while (!Serial);
if (!modem.begin(EU868)){
Serial.println("Failed to start module");
while (1){}
};
Serial.print("Your device EUI is: ");
Serial.println(modem.deviceEUI());

int connected = modem.joinOTAA(appEui, appKey);
if (!connected){
Serial.println("Something went wrong! Retry");
while (1){}
}
modem.minPollInterval(60);

Serial.println("Successfully joined the network!");
pinMode(13,OUTPUT); //Set pin 13 as OUTPUT pin, to send signal to relay
}
void loop() {
// put your main code here, to run repeatedly:
adjust_water_level();
send_level();
for (int i =0;i<10;i++){ /10 * 1 second delay/
check_downlink_messages();
delay(1000);
}
}
void adjust_water_level(){
if (pump_running){
if (water_level < full_level){ /Can't be overfilled/
water_level += 1; /Water level rises/
}
}
else {
if (water_level > empty_level){ /Can't be less than empty/
water_level -= 1; /Water level drops/
}
}
return;
}

void send_level(){
int err;
//modem.setPort(4);
Serial.print("Sending water level: ");
Serial.println(water_level);
modem.beginPacket();
modem.write(water_level);
err = modem.endPacket(true);
if (err > 0) {
Serial.println("Message sent correctly!");
}
else {
Serial.println("Error sending message :(");
}
delay(1000 * 60);
return;
}

void check_downlink_messages(){
if (!modem.available()) { /If there was no message, get out of function/
Serial.println("No downlink message received at this time.");
return;
}
else { /If there is a message/
Serial.print("Message Received: ");
if (modem.read()!= 0){ /The first byte is non-zero/
Serial.println("Starting the pump");
pump_running = true;
digitalWrite(13,HIGH); //if pump is not running, send HIGH value to relay
}
else{
Serial.println("Stopping the pump");
pump_running = false;
digitalWrite(13,LOW); //if pump is not running, send LOW value to relay
}
}
delay(400);
return;
}
The message gets uploaded to the TTN, however I get Payload: {ledState: null} 02 00 00 00 when I send an integer with a value of 2 to the network using the modem.write() function. I did this just for testing purposes. Later I used a uplink simulation to see if perhaps I used the wrong payload formatting, however that is not the case. I should be getting a Payload similar to the simulation, that being Payload: {port: 4, water_level=2).
[image]
Note that I am completely new to both arduino and lora, and this is a simple school project with a deadline, so I would appreciate if someone could help me out ASAP. If it is problematic that this post exists here, I am sorry in advanced, but after TTN v3 came out I am not sure if any older online resources are helpful anymore and I desperately need help.

per1234, 2021-09-02:
Hi @OcimaSagledam. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there:

https://forum.arduino.cc/

closed 2021-09-02 by per1234

GusMartinez, 2022-12-09:
Hi @OcimaSagledam did you manage to solve this? I'm currently facing the same problem and I've couldn't find any solution.
If you know anything that could help, please send me an e-mail gustavo96mm@gmail.com