arduino-libraries/MKRWAN #107: Feature Request: Allow the MKRWAN library to send and receive very simple LoRa peer-to-peer commands

hpssjellis, 2021-11-09:
[image]

Both the  [Arduino MKR WAN 1300 (LoRa connectivity)](https://store-usa.arduino.cc/products/arduino-mkr-wan-1300-lora-connectivity)   and the [Arduino Portenta Vision Shield - LoRa®](https://store-usa.arduino.cc/products/arduino-portenta-vision-shield-lora%C2%AE) are advertised as LoRa connectivity, but they aren't LoRa, they are actually the more advanced LoRaWan connectivity. To clear up some of the confusion for present and future customers, I suggest a Feature Request to add very simple 10 Byte  both way un-encrypted peer-to-peer communication to the official MKRWAN library.

Many businesses could use a very simple DIY LoRa message system that goes > 400 m, which can't be done by WiFi or BlueTooth, also Ethernet needs a cable and Cellular has a monthly cost.

We know the MKRWAN Murata module can do peer-to-peer LoRa connectivity demonstrated by this solved issue using the [GrumpyOldPizza STM3210 Github](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0)  to control the murata module as an installed board see [issue 87](https://github.com/arduino-libraries/MKRWAN/issues/87)    some more information here at [mbed-issue-211](https://github.com/arduino/ArduinoCore-mbed/issues/211)

Most companies will decide to use the multiple LoRaWan connectivity methods ranging from [free and paid TTN](https://www.thethingsnetwork.org/map) to the [free to try cryptocurrency Helium HNT Peoples Network](https://explorer.helium.com/coverage) . These solutions are not as simple as they seem to be and involve lots of steps communicating with about 2 cloud services often including one MQTT server. Each cloud service has it's own setup issues, along with different frequencies for different Continents. 

To get started with LoRa connectivity  a simple peer-to-peer MKRWAN solution would be very useful and also aligned with the Arduino advertising.

[image]

cturvey, 2021-11-09:
Are you talking about some formal, addressable peer-to-peer implementation, with retries and acknowledgement or a raw point-to-multipoint broadcast into the ether? 

If the former, how involved to you see that being? Number of stations? Meshing?

Why stop at 10 bytes? The threshold is frequently trimmed to 64-bytes, but the SX1276/Murata Module is more than capable of pushing 250+ bytes over 800 m at the highest bandwidth settings with minimal effort, 5-6 km with relatively little effort, and certainly beyond 14 km LoS for smaller packets in the lower bandwidth modes. And there are people in better terrain who gotten signals 27+ km, and to balloons on the edge of space.

RAK Wireless has a LoRaP2P mode in the RAK-811, REYAX also has a mode for their modules, as I recall. Perhaps look at those as model, or protocols which could be integrated/supported.

ST has buildable LoRa and LoRaWAN code for these Murata modules, and code generated there is injectable via the existing MKRWAN firmware update tool. And relatively interchangeable with the Discovery, Grasshopper, 1300, etc.
https://www.st.com/en/embedded-software/i-cube-lrwan.html

Implemented 3DR/HM-TRP equivalent telemetry radios here using the the MKRWAN 1300 as a cheap COTS board, with enough pins and MCU horse-power to do a lot additional things.

See also
https://www.haystacktechnologies.com/haystack-for-lora/

hpssjellis, 2021-11-09:
@cturvey Great reply!

Most Makers are often stumped without a starting example that works for their specific board. I would be fine with sending 1 byte, but 10 bytes is more useful. Expanding to a maximum should be shown as easy to do.  My request would be for 2 examples. One, a send to all available LoRa receivers (into the Ether). The other, a bit more complex as a send and receive a reply, no encryption, easy to do. 

Other examples showing how to connect to the above mentioned code would be very useful, but it is not Arduino code and as such is probably too advanced for most Makers.

I have lots of examples using the GrumpyOldPizza technique [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/research/LoRaWan/grumpyOldPizza/finally), but that is fairly hard for my students to install and work with. Not Arduino Easy.

For my specific situation I have a [EdgeImpulse](https://www.edgeimpulse.com/) Machine Learning Vision example steering a toy car using the Portenta with Vision Shield and I would like some simple connectivity just to send an emergency stop, I have tried WiFi and BLE but both seem to have issues as the M7 core is fully used. Now trying LoRa.

[image]

hpssjellis, 2021-11-14:
@cturvey So I can confirm that the [GrumpyOldPizza](https://github.com/arduino-libraries/MKRWAN/issues/87#issuecomment-793530383) code converted by Arduino coder @facchinm does run parallel to a complex vision machine learning program on the PortentaH7 M7 core, with no perceptual speed slow down, whereas presently I can't get any WiFi or BLE code working at all on the M7 core while the EdgeImpulse ML code is running.

My research repo is [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/research/LoRaWan/grumpyOldPizza/finally/lora-nov-2021) with some code snippets, including a basic send and receive 10 byte LoRa code.

If the MKRWAN library could run basic LoRa this could really help writing machine learning code with Arduino boards that has the ability to communicate the ML classifications.

hpssjellis, 2021-12-06:
Can someone show a very simple MKRWAN example that directly uses an ```AT Command```?

I think this is a list of the AT commands [here](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/db/8c/4f/5a/95/9e/40/69/DM00346311/files/DM00346311.pdf/jcr:content/translations/en.DM00346311.pdf)

hpssjellis, 2022-01-04:
We can close this. I have examples on my "Portenta Pro Community Solutions" library here

[https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot3-portenta-vision-shields/dot34-pure-lora](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot3-portenta-vision-shields/dot34-pure-lora)

closed 2022-01-04 by hpssjellis