arduino-libraries/MKRWAN #109: Official AT Commands for firmwares?

reyes73a, 2021-12-07:
Does anyone know where can I get the official docs for the AT commands on the Murata chip? Each firmware has a different set of commands and there is no documentation available on this other than reverse engineering the example codes. However there is a lot of mismatch between provided firmwares and MKRWAN versions...

hpssjellis, 2021-12-18:
https://github.com/arduino-libraries/MKRWAN/issues/107#issuecomment-987329703

flhofer, 2022-01-30:
Hi @reyes73a ,
The actual firmware now is a _not so recent_ fork of the ST I-CUBE-LRWAN modem software. Thus the documentation provided is not up-to-date. Especially as I recently did some major changes to make a lot of stuff actually work.

If you want the list of supported commands, all you have to reverse-engineer is [this file](https://github.com/flhofer/mkrwan1300-fw/blob/master/Projects/Multi/Applications/LoRa/AT_Slave/src/at.c). (Put my copy of the latest software)..

Feel free to flash the firmware provided in the pull request as it really solves a lot of issues. (Can also be done with the MKRWAN standalone sketch, in my pull-request here around)
Hope this helps.