arduino-libraries/MKRWAN #120: More recent LoRaWAN firmware for Murata

janakj, 2022-08-19:
If you need more recent LoRaWAN firmware for the Murata TypeABZ module,  I have been developing an open source replacement for Murata's firmware in [this Github repository](https://github.com/hardwario/lora-modem-abz). It provides an [AT command interface](https://github.com/hardwario/lora-modem-abz/wiki/AT-Command-Interface) backward compatible with Murata's firmware, supports LoRaWAN 1.0/1.0.4 and regional parameters 1.0.3,  and it is based on the most recent version of [LoRaMac-node](https://github.com/Lora-net/LoRaMac-node), Semtech's LoRaWAN reference implementation.

I apologize for posting this as an issue, but I haven't found any other way to reach out.

per1234, 2022-08-22:
Thanks for your work on this interesting project, and for sharing it @janakj!

If I understand correctly, this is not intended to propose any actions within the MKRWAN library code base or documentation. For that reason, I will close the issue.

It will continue to serve its purpose of bringing this alternative firmware to the attention of interested parties as a permanent item in the repository's issue tracker.

closed 2022-08-22 by per1234

janakj, 2022-08-22:
Hi @per1234, that's correct. I am not proposing any specific action for MKRWAN. I just wanted to let you know that an alternative open-source firmware for the Murata modem exists, in case anybody is looking for alternatives. I just noticed #64, maybe I should have added a comment there instead.

I quickly scanned the list of open issues in [mkrwan1300-fw](https://github.com/arduino/mkrwan1300-fw) and it would seem that most of them would be fixed in lora-modem-abz.