arduino-libraries/MKRWAN #127: how to joinOTAA with different datarate

arnschi, 2023-08-21:
Hi.

I am using MKRWAN with the standard arduino IoT library. My device is quite on the edge regarding signal quality, and found out that joinOTAA uses a quite weak data rate ("bandwidth": 125000, "spreading_factor": 7, "coding_rate": "4/5")

In the rare cases where the join is successful. the data packets are sent with a better spreading factor of 12.

This also can be seen in practical use. For joining, the antenna has to be outside of the window, but when the device is joined, the connection is stable indoors without problems.

How can I make the join.OTAA be done with a better spreading factor? I tried to modify the Arduino_LoRaConnectionHandler.cpp by adding a call to dataRate(0) into the handleInit. But this has no effect, TTN console still tells me that the join-request packets are sent with SF7

It seems to be a known problem, at least for the mkrwan firmware. But is there really no solution to join the network with the highest range which is possible?

regards,
Arnschi

per1234, 2023-08-22:
Hi @arnschi. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2023-08-22 by per1234

arnschi, 2023-08-30:
Hi @per1234  - thanks for reading. In principle I agree, but in some way this is related specific to the project. It happens with the firmware provided in MKRWAN and the API dataRate() seems to have no effect, at least when connecting.

I'll give it a try in the arduino forum, however it seems the arduino forum is almost dead for MKRWAN device.