arduino-libraries/MKRWAN #140: No way to force US Sub-band #2 (channels 8-15) to join Helium network reliably

Joshbloam, 2024-12-22:
Hello. I have spent hours trying to find a way to force the modem to only send joins on US915 Sub-band #2 (channels 8-15) used by Helium. I can force the modem to use sub-band #1 by setting modem.begin(US915_HYBRID) and talk to The Things Stack successfully. The problem is Heliun uses sub-band #2 and I need a way to enforce joins only use US915 channels 8-15. 

Does anyone know how to do this? My fear is there is no way to do this currently. Could someone look at creating a variation of US915_HYBRID which only uses US915 band #2 (Helium) instead of #1 (TTS)?

Currently I can get successful joins one out of 50 attempts it seems. Let me know I can post the entire sketch if needed. Thanks! Josh

`if (!modem.begin(US915)) {
    Serial.println("Failed to start module");
    while (1) {}
  };
  modem.disableChannel(0);
  modem.enableChannel(1);    // only one enabled for Helium
  modem.disableChannel(2);
  modem.disableChannel(3);
  modem.disableChannel(4);
  modem.disableChannel(5);
  modem.disableChannel(6);
  
  modem.configureClass(CLASS_C);
    int connected = modem.joinOTAA(appEui, appKey);
 while (!connected) {
    Serial.println("Something went wrong; are you indoor? retring after 30sec");
    delay(1000); 
    int connected = modem.joinOTAA(appEui, appKey);
  }
   
  
  
  `

Joshbloam, 2024-12-27:
Never mind. I figured out how to set a custom channel mask for US915 Helium!

Deiikerc, 2024-12-30:
Hello, could you tell me how you configured custom channels for US915? I have an SX1302 that I'm using with the packet forwarder and it seems that the node (initially) has problems transmitting data to the gateway.

Joshbloam, 2024-12-30:
Looking through the source I discovered a sendmask method. I set the reginal band to US915 and apply the mask. 
Through trial and error this seems to keep joins on sub-band2 channels 8-15 plus channel 65 only.

modem.begin(US915)
modem.sendMask("ff0000000000000080020000");

Deiikerc, 2025-01-02:
> Looking through the source I discovered a sendmask method. I set the reginal band to US915 and apply the mask. Through trial and error this seems to keep joins on sub-band2 channels 8-15 plus channel 65 only.
> 
> modem.begin(US915) modem.sendMask("ff0000000000000080020000");

I'm really grateful, I was having a lot of trouble getting my gateway to listen to this node.

Joshbloam, 2025-01-02:
> I'm really grateful, I was having a lot of trouble getting my gateway to listen to this node.

Glad you had success!